victronenergy/venus #190: timezone in Venus vs in VRM

mpvader, 2017-09-07:
In VRM, users can set a time zone. In Venus too, which is confusion and leads to problems, since on VRM it is also related to lat long.

The best would be that users configure it Venus, which is then also used on VRM. And then obviously it won't be settable in VRM, at least for sites that send the time zone (data attribute `tz`).

Which in itself is fine, but does pose a migration issue: lots of Venus devices don't have their Timezone configured. So just adding will So suddenly overwriting all configured timezones in the VRM database with /UTC 

In Venus, the timezone is set on the data partition, in `/data/conf/settings.xml, <TimeZone>`. Looks like there is no way, at this moment, to distinguish between `not set` and `purposefully set to UTC`.

On VRM:
- if there is timezone in lastLogData <> UTC, use that. Otherwise use the user setted time zone.
- only do timezone zone setting in VRM frontend when there is no timezone record in lastlogdata. If there is, show a note that timezone needs to be configured on the Venus device itself.

@thlassche, what do you think?

thlassche, 2017-10-03:
@mpvader Sounds goud, I'll have Alex implement it

mpvader, 2017-10-03:
@wiebeytec fyi

AlexImb, 2017-10-04:
@mpvader do you know why some devices return the timezone in the Continent/Country format and other only the Continent and a trailing slash (Europe/, also in the picture)? In this case to which of the timezones should be mapped because there are about 4 timezones per continent? This case should be treated in same way as the /UTC? 

Is there any specific reason why the UTC has a leading slash? Does it mean something?

[image]

Thank you!

mpvader, 2017-10-04:
Hi @AlexImb no I don't. Can you relate it to the type of gateway? Are there many with the continents without country?

I looked a few up from your list, on the vrm diagnostics page, and I then see a proper time zone. For example site 1507 has Europe/London. Not Europe/. Are you sure that the list is correct?

mpvader, 2017-10-04:
> leading slash on (only) /UTC

I don't know why there is a leading slash for only UTC. Are there also UTCs in this field without leading slashes? The string comes from Linux.

jhofstee, 2017-10-04:
the leading / just means there is no continent associated with it.

thlassche, 2017-10-19:
This is implemented now. However, there is some discrepancy between the timezones PHP knows and the ones from UNIX, which I just found out. If we encounter one PHP does not know, we fall back to the timezone column in the sites table for now. 

These are the ones PHP knows:
```
array(425) {
  [0]=>
  string(14) "Africa/Abidjan"
  [1]=>
  string(12) "Africa/Accra"
  [2]=>
  string(18) "Africa/Addis_Ababa"
  [3]=>
  string(14) "Africa/Algiers"
  [4]=>
  string(13) "Africa/Asmara"
  [5]=>
  string(13) "Africa/Bamako"
  [6]=>
  string(13) "Africa/Bangui"
  [7]=>
  string(13) "Africa/Banjul"
  [8]=>
  string(13) "Africa/Bissau"
  [9]=>
  string(15) "Africa/Blantyre"
  [10]=>
  string(18) "Africa/Brazzaville"
  [11]=>
  string(16) "Africa/Bujumbura"
  [12]=>
  string(12) "Africa/Cairo"
  [13]=>
  string(17) "Africa/Casablanca"
  [14]=>
  string(12) "Africa/Ceuta"
  [15]=>
  string(14) "Africa/Conakry"
  [16]=>
  string(12) "Africa/Dakar"
  [17]=>
  string(20) "Africa/Dar_es_Salaam"
  [18]=>
  string(15) "Africa/Djibouti"
  [19]=>
  string(13) "Africa/Douala"
  [20]=>
  string(15) "Africa/El_Aaiun"
  [21]=>
  string(15) "Africa/Freetown"
  [22]=>
  string(15) "Africa/Gaborone"
  [23]=>
  string(13) "Africa/Harare"
  [24]=>
  string(19) "Africa/Johannesburg"
  [25]=>
  string(11) "Africa/Juba"
  [26]=>
  string(14) "Africa/Kampala"
  [27]=>
  string(15) "Africa/Khartoum"
  [28]=>
  string(13) "Africa/Kigali"
  [29]=>
  string(15) "Africa/Kinshasa"
  [30]=>
  string(12) "Africa/Lagos"
  [31]=>
  string(17) "Africa/Libreville"
  [32]=>
[... 770 lines cut ...]
  string(17) "Pacific/Rarotonga"
  [418]=>
  string(14) "Pacific/Saipan"
  [419]=>
  string(14) "Pacific/Tahiti"
  [420]=>
  string(14) "Pacific/Tarawa"
  [421]=>
  string(17) "Pacific/Tongatapu"
  [422]=>
  string(12) "Pacific/Wake"
  [423]=>
  string(14) "Pacific/Wallis"
  [424]=>
  string(3) "UTC"
}
```

I added a small snippet to fix the most common errors:

```
            // Asia/Atlantic/Reykjavik --> Atlantic/Reykjavik
            while(count($arrTimezone) > 2)
                array_shift($arrTimezone);
```

mpvader, 2018-07-12:
hey @thlassche I really would like to get this finished; migrating Venus to use the PHP timezone database is no option 😄 , so.. pls fix it and hopefully in a way that I don't have to be involved or worry about it?

So ,perhaps just make PHP work with tzdata instead of whatever it is working on now?

thlassche, 2018-07-12:
@mpvader Ok, this is a while ago, let me look into it, and I'll do me best so that you don't have to worry about it ;)

thlassche, 2018-07-27:
This one is fixed by also including BC-compatible timezones in PHP. They now all match (except `Africa/UTC`, which defaults to `UTC` which is fine)

mpvader, 2018-07-27:
Can the issue be closed / is the Timezone sync now implemented?

mpvader, 2018-07-27:
Dont forget this:

https://www.victronenergy.com/live/vrm_portal:change_log

closed 2018-07-27 by mpvader