victronenergy/venus #191: gridless octo: test bluetooth

mpvader, 2017-09-12:
Its hard for this to not work, but before ordering the next PCB version it needs to be tested.

@mansr pls test this.

WiFi has been tested by Unipoin.

mansr, 2017-09-19:
The built-in bluetooth is miserable in the same way as on Venus GX. With a USB bluetooth dongle it works fine.

mpvader, 2017-09-19:
@mkrenzke miserable means that it has miserable Bluetooth reception like the Venus GX.

I hoped/expected it to be good, since you also reported that the wifi reception on the Gridless Octo is good. Unlike wifi on the Venus GX. Any ideas or comments?

mkrenzke, 2017-09-19:
That's bad news. I had the same expectations.

@fhuebner is already looking into it.

mkrenzke, 2017-09-19:
@mansr: How do you test bluetooth connectivity?

mansr, 2017-09-20:
A basic check of reception can be done with these commands:
```
hciconfig hci0 up
hcitool -i hci0 lescan
```
If there are any devices transmitting nearby, their MAC addresses and names are printed.

mkrenzke, 2017-09-20:
@mansr: Thank you!

With `btmon` we see signal strength of -74 dB even for devices in 10 meters distance with the built-in module. Which is not too bad.

Which antenna do you use? The one we sent or any other?
What is your testing scenario?

mansr, 2017-09-20:
I didn't receive an antenna, so I tried with a couple I had lying around. The transmitting device is less than a metre away, so I'd expect it to work even without a proper antenna.

mkrenzke, 2017-09-20:
When I scan for BLE networks (with advertising on) with my mobile phone I get -35 dB in 1 meter.

[image]

Without antenna it is about -80 dB in 1 meter.

mpvader, 2017-10-16:
discussion is continued here https://github.com/victronenergy/venus/issues/207

closed 2017-10-16 by mpvader