victronenergy/venus #193: vup: add option to retrieve raw firmware version

mpvader, 2017-09-13:
Relates to https://github.com/victronenergy/venus-private/issues/32.

See below for current output of vup. An option will be added to get the unformatted firmware version instead.

```
# vup -x -vv

<vuprun version="2.0.3">
<message type="debug" module="canhw_drv" timestamp="2">scanning /usr/local/lib/canhw</message>
<message type="debug" module="canhw_drv" timestamp="3">found socketcan</message>
<message type="debug" module="canhw_drv" timestamp="4">loading '/usr/local/lib/canhw/canhw_socketcan.so'</message>
<message type="debug" module="canhw_drv" timestamp="6">locating veCanhwRegister</message>
<message type="debug" module="canhw_drv" timestamp="7">adding driver socketcan</message>
<message type="debug" module="skcan" timestamp="9">ignoring lo</message>
<message type="debug" module="skcan" timestamp="10">ignoring sit0</message>
<message type="debug" module="skcan" timestamp="11">ignoring eth0</message>
<message type="debug" module="skcan" timestamp="11">ignoring lo</message>
<message type="debug" module="skcan" timestamp="12">ignoring eth0</message>
<message type="debug" module="skcan" timestamp="13">ignoring lo</message>
<message type="debug" module="skcan" timestamp="14">ignoring eth0</message>
<message type="debug" module="task" timestamp="14">enumerate done</message>
<message type="debug" module="" timestamp="15">Using can0</message>
<message type="normal">Querying devices</message>
<discovered>
<device nad="0x14" man="358" serial="0013556 HQ1229VKPF9" boot="1.00" mode="app" updatable="True" id="101" description="Skylla-i Control" version="2.01" settingsMigration="True">
</device>
<device nad="0x15" man="358" serial="0013631 HQ1229ZVH4Y" boot="1.00" mode="app" updatable="True" id="101" description="Skylla-i Control" version="2.01" settingsMigration="True">
</device>
<device nad="0x21" man="358" serial="0015639 HQ12424SLBZ" boot="1.00" mode="app" updatable="True" id="106" description="Skylla-i 24/100 (3)" version="c2.05" settingsMigration="True">
</device>
<device nad="0x22" man="358" serial="0015644 HQ1242WUBN3" boot="1.00" mode="app" updatable="True" id="106" description="Skylla-i 24/100 (3)" version="c2.05" settingsMigration="True">
</device>
<device nad="0x24" man="358" serial="" boot="1.01" mode="boot" updatable="True" id="108" description="BlueSolar Charger MPPT 150/70" settingsMigration="False">
</device>
<device nad="0x25" man="358" serial="0016252 HQ1250JFX2W" boot="1.01" mode="app" updatable="True" id="108" description="BlueSolar Charger MPPT 150/70" version="2.05" settingsMigration="True">
</device>
</discovered>
<message type="debug" module="task" timestamp="2042">done in 002.014</message>
</vuprun>
```

mpvader, 2017-10-16:
In Venus v2.12~1

jhofstee, 2017-10-17:
second attempt, included in v2.12~2 (building now)

closed 2017-10-17 by jhofstee