victronenergy/venus #195: "Control without grid meter" non efective if grid meter is connected

wilhoj, 2017-09-19:
I have done multiple experiments and have found out that in ESS configuration system is always following readings from grid meter, not internal AC out readings if grid meter is connected to system.

Firmware v2.09

Components of the system:
Venus GX
Multigrid 3000
BlueSolar MPPT 150/35
Energy meter ET340
RS485 to USB interface
AGM battery bank and solar panels

vrm installation id 14424

Thing is why i want to use "Control without grid meter" is that you could not limit power drawn from battery atm.. if you limit interter power you are also efectively limiting pv power too. I have steady few hundreds of watts load on AC out, but on grid i have big usage spikes of water heater etc which will cause huge current draw from battery if there is no solar power available.

edevries-victron, 2017-09-19:
The 'control without grid meter' option allows systems without grid meter (where all loads are connected to the AC-out) to be used as ESS system. If a compatible grid meter is detected, the system will also compensate for loads between the grid meter and the Multi.

At this point the only way to get the behavior you want is to disconnect the meter. To prevent current spikes drawn from the battery, you could set the 'maximum inverter power' in the ESS settings on the CCGX. The system will still try to compensate for the loads between grid meter and Multi, but the battery discharge current will be limited.

wilhoj, 2017-09-19:
OK, is this feature on roadmap of upcoming versions?

isnt that the case that 'maximum inverter power' will also limit power inverterd from solarpanels?

mpvader, 2017-09-20:
Hi @wilhoj 

> OK, is this feature on roadmap of upcoming versions?

Making the 'Control without grid meter' setting work even when there is a grid meter connected you mean? Right now its not.

> Isn't that the case that 'maximum inverter power' will also limit power inverterd from solarpanels?

Correct, that is the disadvantage of that setting. The good news is that we are working on a number of changes which might also make it possible to have a real max battery discharge current setting. I don't know for sure yet if there really will be such a setting, it be clearer in the coming months.

Wholistic, 2017-11-25:
I am not clear why you have the grid meter if you do not want the multi to push power out to run the loads that are there, but not on the AC-Out. 

If it is just to collect the data, you can change the setting in the Energy Meter menu of the CCGX from "Grid Meter" to "PV inverter" and then turn on Control without grid meter. You will also need to reconfigure the grid meter, as the wiring configuration is different in this mode. 

I am currently preparing more diagrams to explain all the possible configurations for how the grid meter can be used and configured to achieve the desired result. There are many options.

wilhoj, 2017-11-25:
Hi, 

That was mainly because that batteries i did have on that time was not able to handle those usage spikes example water heater was causing.. and if i did limit inverter power it was efectively limiting inverted solar energy too.. simplest solution would be using battery inverted power based on ac out and allow surplus solar energy feed to grid.

You are right that in that configuration ac meter would be pretty much only for information.. 

Solution you recommended is ok, tho bit counter intuitive and needs rewiring of the meter...

i am just wondering that would that be pretty simple thing to do in software to allow grid meter to be connected and still act as "Control without grid meter"

i agree that this definitely bit rare need.. and with my current better batteries i dont need such a feature..

Wholistic, 2017-11-26:
I too would like more ‘flexibility’ in how the information from the energy meter is interpreted and used by the CCGX. However given that most configurations that I have required are currently possible with physical wiring and current software settings I don’t see a big demand. 

The possible energy meter configurations do need further documentation, it is a bit of trial and error at the moment. Even though most info is out there somewhere it isn’t clearly diagrammed and described in the one spot.

wilhoj, 2017-11-26:
I agree,

One other thing is bit confusing, that "Control without grid menter" option is still there and enabled, although it wont do anything if grid meter is connected.. at least you could disable this option if there is gridmeter connected.

mpvader, 2019-06-22:
since above discussion, we've reworded the item:

[image]

I understand that this doesn't answer your original request @wilhoj, but still I'm going to close this issue. Even though I understand your request; and it makes sense, nobody else is asking for this; and looks like you don't need it either anymore.

closed 2019-06-22 by mpvader