victronenergy/venus #196: Stop Serial-starter service

PaulB2Code, 2017-09-19:
Dear Venus team,

We would like to stop the serial starter service properly to use the usb for an other use than a GPS. 

How could we do that?

Thank you 

Best

mpvader, 2017-09-19:
@PaulB2Code this is for a SIM800, right?

I logged into your system, and found these devices:

```
root@beaglebone:~# lsusb
Bus 002 Device 004: ID 0bda:b720 Realtek Semiconductor Corp. 
Bus 002 Device 005: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Is the SIM800 connected via a serial-usb converter from Prolific (ie the PL2303 port)? I thought the SIM800 could be direct connected to USB, and was hoping for some more specific identification on which the serial-starter could decide to skip that tty port.

Last question, it looks like the hardware is a Victron Venus GX, please confirm.

mpvader, 2017-09-19:
And, while logged in, herewith a copy of the usb details of the connected prolific:

```
# lsusb -D /dev/bus/usb/002/005
Device: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial Port
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x067b Prolific Technology, Inc.
  idProduct          0x2303 PL2303 Serial Port
  bcdDevice            3.00
  iManufacturer           1 Prolific Technology Inc.
  iProduct                2 USB-Serial Controller
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x000a  1x 10 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

PaulB2Code, 2017-09-19:
Yes, we use a cable to convert usb to uart as this following cable.

The uart speed for our sim is 115 200. 

[image]

The hardware is a venus GX with a multiplus connected.

mpvader, 2017-09-19:
Why not connect the SIM800 via USB? That would probably also solve the multiple parallel channel issue (being able to send/receive text messages ar the same time as having an internet connection).

PaulB2Code, 2017-09-19:
The sim can today communicate only with uart and the issue is that multiple services want to use the same serial port. 
So TX and RX info are not shared by the two services. 
This why we wanted to stop serial services.

mkrenzke, 2017-09-19:
As far as I know the SIM800 does have USB support as debug console only (pins 24 & 25)

http://simcomm2m.com/En/module/detail.aspx?id=138

mansr, 2017-09-20:
The serial-starter script uses the product name (iProduct in the listing above) of USB devices to determine which service(s) to start. The mapping is defined in the `get_program` function in `serial-starter.sh`. For unknown products (matched by the * wildcard), the gps and vedirect services are attempted. To have devices like yours left alone, you can modify this function to return `ignore` for a product name of `USB-Serial_Controller`.

mpvader, 2017-09-20:
@PaulB2Code use above answer from @mansr to get it working on your desk / during development.

The proper solution would be make the serial-starter detect that it is a modem, or even more specifically a sim800 modem. One way would be actively probing: sending `AT` or similar. But that would be rather messy, to send out commands to a device of which you don't know its type, nor its baudrate. The real solution for the uart connected SIM800 would be to use a usb-serial device and flash that with another product name. Like we do in the VE.Direct USB cables: in them there is a ftdi FT230X. Of which we change the USB identifiers with the [FT_PROG tool](http://www.ftdichip.com/Support/Utilities.htm#FT_PROG) in the manufacturing process.

@PaulB2Code agree with above? Or you see another solution?

mpvader, 2017-09-20:
Fyi all, @mkrenzke is working on modem support in Venus as well, but then 3G. The chosen module is the SIM5360, which will be connected via USB. Next step is to receive an evaluation board (later this week), and test that it can be powered via USB. Rather than requiring its own power supply.

PaulB2Code, 2017-09-24:
Hi All, 
Thanks for the replies. 
We will try this solutions. 
I will keep you informed, if we get stuck.

mpvader, 2017-10-16:
Hi @PaulB2Code I'll close this issue now. Welcome to re-open or make a new issue if you get stuck again.

closed 2017-10-16 by mpvader