victronenergy/venus #199: wind-mppt-bornay: add to serial-starter

mpvader, 2017-10-02:
Basics:
- the wind controller talk ModbusRTU over RS485
- Bornay has implemented a Python script that talks to the controller and publishes the data on D-Bus

Changes required before adding the script to the serial-starter:
- the script should accept a serial port as a command line argument (and then it should not scan all the ports)
- the script should quit in case it doesn't find a wind controller on the RS485

CarlosBornay, 2017-10-06:
Hi Matthijs, we have develop the command line argument to accept a serial port and we remove the port scan.We ran the script to test the new code and works fine.Also the script stops if any error happend.  We also check the d-bus paths as Jesus told us and also the script checks if the process is te correcto to starts de Dbus. We think that with this changes all is done by our side until now.  Here you have the link: 

https://github.com/CarlosBornay/Bornay-venus-driver

mpvader, 2017-10-09:
@jepefe pls review

izak, 2017-10-12:
So far from the review.

It seems we can significantly reduce the dependencies by packaging our own pymodbus and removing the `twisted` requirement. This is an old issue I've run into before, you don't need twisted to use pymodbus, you only need it of you're going to use the asynchronous client features. We should definitely do this if we're going to package it for Venus. It's easy, we just ignore the dependency and package python-pymodbus without it.

I've also reshaped things a bit, added a `setup.py` file and put things into a namespace. It also adds a stub to the `bin` directory so you can call `bornay` directly.

Also added the velib_python as a git submodule as we do with other things. The dependencies are listed in `setup.py`, so we avoid having to drag a zip file with dependencies along. You can install the dependencies from pypi.

We'll need to add recipes for: 1) pymodbus, 2) pyserial, and 3) six.

The branch with changes can be installed with pip, or setup.py, eg:

    pip install /path/to/driver

or

    python setup.py install

I usually prefer working in a virtual environment, eg:

    virtualenv venv
    venv/bin/pip install -e /path/to/driver
    venv/bin/bornay

I pushed the changes [here][ib].

[ib]: https://github.com/izak/Bornay-venus-driver

izak, 2017-10-13:
I deleted my previous comment because I changed my mind on how to do it.

Here are some things I propose you change:

1. We add `velib_python` as a submodule. I've already done that [here](https://github.com/izak/Bornay-venus-driver)
2. Line 203, `self.dbusservice = VeDbusService('com.victronenergy.windcharger.bornay_ttyUSB0')`, that should use the serial port as passed on the commandline.
3. Shorten the command line option, make it just `--serial`.

For deployment, I will then:

1. Create a recipe for packaging pymodbus. We already have recipes in OE for pyserial and six.
2. Create a recipe called `dbus-bornay-windplus`.

Then finally we just need to add it into the serial-starter loop.

izak, 2017-10-17:
I see code that attempts to change the port permissions on Raspbian, when running non-root. The correct solution is to add the `pi` user to the `dialout` group, as the `dialout` group always has write permission to serial ports. Run this command once, and you never need to `chmod` the port again.

    adduser pi dialout

I'm going to kill the `systemtype` code, it should not be necessary once you do this. You also don't need to do this on Venus, it can already read the serial port.

izak, 2017-10-17:
@CarlosBornay I made a significant amount of changes. Please see the list [here][ib]. Please test that it still works for you.

On your development platform, which I deduce from the code is Raspbian, add the user to the dialout group, then log out and back in, and make sure it still works. That is a cleaner solution than calling sudo and chmod from inside the code.

Also, on Raspbian, you should be able to install the required modules using `apt-get install python-pymodbus`, that should compensate for the fact that I removed the ext.zip binary blob. Then you no longer need the external modules in that directory.

Finally, I added [recipes][venus] so this can be pulled into Venus.

To add this into serial_starter, edit `/opt/victronenergy/serial-starter/serial-starter.sh` and add in the relevant place:

   svc_bornay=dbus-bornay-windplus

Then in `get_program`, put bornay into the list:
```
        FT232R_USB_UART)                 
            echo cgwacs:gps:vedirect:bornay     
            ;;                           
        USB-RS485_Cable)                                 
            echo cgwacs:redflow:bornay          
            ;;                           
        USB-Serial_Controller_D)    
            echo cgwacs:gps:bornay             
            ;;                      

Test if that works to auto-detect it.

[ib]: https://github.com/izak/Bornay-venus-driver/commits/master
[venus]: https://github.com/izak/meta-victronenergy/commits/feature/bornay-mppt

CarlosBornay, 2017-11-02:
Dear Izak, 
We have tested the script and it works well. In this case we cannot test it in the Color Control because we had cannot install these libraries. We have found problems to install needed libraries. If you consider that is necessary to test all here please give us instructions or connect with us via Teamviewer for example to explain it. But he have tested at the raspberry and the communications works.

mpvader, 2017-11-02:
@izak pls add those changes you made (meta-victronenergy) to the todo list for inclusion in master.

@CarlosBornay that will take care of the missing package for you.

izak, 2017-11-03:
Hi @CarlosBornay 

I just created a [pull request][pr] so that this driver can live in your github account. Once merged, please tag the release (`git tag -a v1.5.1 -m v1.5.1`) and reply here so I can merge the recipes into venus.

[pr]: https://github.com/CarlosBornay/Bornay-venus-driver/pull/1

jhofstee, 2017-12-05:
seems this is done, https://github.com/CarlosBornay/Bornay-venus-driver/releases

mpvader, 2018-03-13:
The driver code is now hosted here: https://github.com/victronenergy/dbus-bornay-windplus.

The latest release, v1.5.4, is in Venus since v2.20~11.

Next step is for Bornay to test if it now works properly with a Victron RS485 cable. And if it does we can add it to the serial-starter.

mpvader, 2020-04-23:
Bornay was put in touch with Ivo to help them further as a "venus-consultant". Closing this.

closed 2020-04-23 by mpvader