victronenergy/venus #202: ESS: loads on AC-out not fed from CAN-bus battery when it is fully charged

edevries-victron, 2017-10-04:
This problem occurs when:
* CAN-bus battery reports a max charge current of 0A
* There are loads on AC-out
* The ESS grid setpoint setting exceeds 30W (default value is 50W since CCGX v2.07).

In this situation, the /Hub4/DisableCharge flag on the vebus D-Bus service will be set. This will not only cause charging to stop, it will also put the system in passthru mode, which also disables the inverter.
Loads on AC-out will then be supplied with grid power.

edevries-victron, 2017-10-04:
To fix this:
* hub4control must ignore max charge current from battery (this is handled by systemcalc)
* do not set the DisableCharge flag any more

mpvader, 2017-10-26:
Fixed in v2.11-bol5.

Also on todo list for v2.12

mpvader, 2017-11-03:
also in v2.12~6 now (backported to hub4control master)

closed 2017-11-03 by mpvader