victronenergy/venus #203: Let user assign names to devices

ThiemoVanEngelen, 2017-10-05:
Typical use case: multiple BMVs (Marine application) or multiple MPPTs.

### Specifics per device

  * BMV-60x cannot store a custom name (but its not necessary to support this device, too old)
  * BMV-70x cannot store a custom name, but needs to be supported (and making a fw update is not easy either, code memory is full)
  * BMV-712 can store a custom name, but vedirect-interface does not support this yet
  * VE.Direct MPPTs cannot store a custom name
  * VE.Direct Inverters cannot (?) store a custom name
  * VE.Can MPPTs can probably store a custom name (NMEA2000 installation name fields)
  * Skylla-is and other VE.Can chargers probably support it (NMEA2000 installation name fields)
  * BMVs connected via VE.Can: most probably the vedirect to vecan interface supports those same NMEA2000 installation name fields)
  * For VE.Bus this is not relevant: we support only one VE.Bus system

A similar project has been done for VictronConnect, by @ThiemoVanEngelen. And there the VE.Direct dongle stores it itself in case the VE.Direct device doesn't support the custom name VREG. It just stores it, its not coupling it to a serial number or something like that.

### Proposal

Since many devices don't support custom names, we'll have to store them on the CCGX. Either only in that case, or simply in all cases.

Then the next choice where to put the logic: the gui or all the different services. Adding it to each and every service is more work, but only adding it to the gui means that it then doesn't work on VRM.

Adding it to vedirect interface is quite simple, several days of work for Thiemo. The name will then be stored in localsettings, linked to the serial number. That some BMV-712s have no serial number is not relevant: for them the name will be stored in the BMV.

A nice to have for the Smart MPPTs is that they can store there names themselves too, so BLE (VictronConnect) and VE.Direct (Venus) use the same name instead of storing two names in the same device.

mpvader, 2017-10-16:
fyi @ThiemoVanEngelen: a cut & paste of a (ancient) page on the internal wiki:

# Assigning a custom product name
On the D-Bus, there already is the path /ProductName. Definition, from [Dbus API spec](https://www.victronenergy.com/live/open_source:ccgx:d-bus: 'As defined in products.c in velib, probably also on VE.Can as PGN 1F014, field 4 ‘Manufacturer’s Model ID’.'. We will leave this as is, it will be used by the Device menu in the CCGX GUI to show the type of product.

A new mandatory object path is added: /CustomName

When possible, this name is stored in the product (in the Lynx Shunt VE.Can for example), or the interface to the product (BMV canbus cable). If that is not possible, it is stored in the CCGX, via localsettings.

## How should it work?
  * When the user has not set a custom name, a GetValue or GetText on /CustomName will reply the same text as /ProductName
  * Max length of the name is 32
    * The strings fields in NMEA2000 PGN 1F014 are all 32 bytes long.

## Example workflow
  - When the user first powers up the CCGX, he sees two times 'BMV-600S' in the product list.
  - He navigates to that device menu of the first, and there sees two entries:
    - Product name: 'BMV-600S' (not editable)
    - Custom name: 'BMV-600S' (editable)
  - He starts editing the Custom name, the editor initializes with 'BMS-600S'.
  - He changes it to 'BMV-600S Service battery'.
  - The GUI does a SetValue with that text to /CustomName/
  - The service will store it where ever he wants (localsettings or the in the product itself)
  - From now on the GUI will show 'BMV-600S Service battery in the product list as well as under Custom name in the device name.
  - vrmlogger will also read the name with GetValue to /Device/CustomName/, and updates the name in the VRM Portal database.

## TODOs
  * Change the GUI to user /CustomName for the list of products, instead of /ProductName
  * Add /CustomName to all services (or at least to the important one: vedirect-interface, since that handles BMVs and MPPTs.
  * Add configuring /CustomName to the 'Device' menu in the GUI. When a user starts editing for the first time, the standard product name is shown as a start. 
  * Add /CustomName to the VRMLogger / VRMDatabase / VRM Portal.
  * Add /CustomName and its rules to [dbus-api spec](https://www.victronenergy.com/live/open_source:ccgx:d-bus)

ThiemoVanEngelen, 2017-10-19:
Implementation is done in vedirect_interface v3.21.

gui implementation is done in branch WIP_CustomName, waiting for merge into master.

Behaviour is a bit different than given above.
/CustomName will contain an empty string when a custom name has not been set. This allows a user of the dbus paths to distinguish the case that a name has been set or not.

ThiemoVanEngelen, 2017-10-19:
[dbus-api spec](https://www.victronenergy.com/live/open_source:ccgx:d-bus) has been updated to include /CustomName.

mpvader, 2017-11-08:
Included in v2.12~8. It now works for all important cases. 

Custom names work for all VE.Direct products:
- where possible (BMV-712) the name is stored in the MPPT
- if there is no serial number (BMV-600), the name is stored in localsettings, by the tty number
- otherwise its stored in localsettings by the serial-number.

It also works for:
- dbus-fronius (ethernet/wifi connected Fronius and SMA PV Inverters)
- cgwacs (gridmeter & pvinverter)
- digital inputs (soon to be included in Venus, see #136 )

It does not work for:
- VE.Bus devices
- VE.Can products
- CANBUS-bms batteries (LG, BYD, etc.)
- PV Inverters measured with the AC Sensor
- any other other thing (qwacs, redflow, & anything else I forget to mention here)

Out of above list, only for the VE.Can products it would be nice to support that some day. But only nice, so closing this issue now.

closed 2017-11-08 by mpvader