victronenergy/venus #204: ESS: Make the state displayed on the GUI tiles more informative

izak, 2017-10-05:
Presently the GUI shows the vebus state of the inverter. This often leads to confusion, because the vebus might be indicate bulk state while the battery is in fact being discharged. Users also have the need to know *why* the Multi might be doing something, for example it might stop discharging because the BMS told it to, or because BatteryLife is active and MinSoc has been reached.

The goal is to communicate the ESS state (when active) to the user, as well as a reason for that state (when applicable).

Identified states:

ESS has broadly four states (and a fifth when the grid is down):

1. Self Consumption
2. Discharge Disabled
3. Slow Charge
4. Sustain
5. Inverting

While in Self Consumption mode, charging or discharging is usually apparent by just looking at the numbers on the screen, or the direction the balls move.

When discharge is disabled, we want to show why. Possibilities:

1. Disabled by BMS
2. Soc is low

When BatteryLife is active and we have a low SoC or we're slow-charging, we want to indicate that BatteryLife is in charge of this decision, possibly by showing some kind of "protection" icon.

izak, 2017-10-12:
Basic design so far. First, I found some `SystemState` info with the present inverter info. I augmented this with some extra states:

```
        SystemStateDescription { value: 0x20; name: "charging"; text: qsTr("Charging")},
        SystemStateDescription { value: 0x21; name: "discharging"; text: qsTr("Discharging")},
        SystemStateDescription { value: 0x22; name: "discharge_disabled"; text: qsTr("Discharged")},
        SystemStateDescription { value: 0x23; name: "slow_charge"; text: qsTr("Slow Charge")},
        SystemStateDescription { value: 0x24; name: "sustain"; text: qsTr("Sustain")},
```

I then added some logic to systemcalc, roughly as follows:

1. Ask the vebus for `/VebusMainState`.
2. If the bus main state is 8 (Passthru):
   * Query `/Bms/AllowToDischarge`, set systemstate to Discharged and reason to `BMS_SAID_NO` if set
   * Query `/BatteryOperationalLimits/MaxDischargeCurrent`, set systemstate to Discharged and reason to `BMS_SAID_NO` if it is zero.
3. If the bus main state is 9 (charging):
   * Query `/Hub4/Sustain` and set systemstate to Discharged, reason to `SUSTAIN` if set
   * Query `/Hub4/AssistantId`, if it exists...
      * Query `/Settings/CGwacs/BatteryLife/State`
         * if state in 2, 3, 4, 10, set systemstate to charging (or discharging, depending on `/Dc/Battery/Power`).
         * if state in 5, 11, set systemstate to discharged, reason to `LOW_SOC` (or `LOW_SC_BATTERYLIFE`)
         * if state in 6, 12, set systemstate to `SLOW_CHARGE`

izak, 2017-10-12:
* Changes to the [gui][gui] code.
* Changes to [systemcalc][sc], just adding a delegate.

[gui]: https://github.com/victronenergy/gui/tree/feature/ess_state_display
[sc]: https://github.com/izak/dbus-systemcalc-py/tree/feature/systemstate

edevries-victron, 2017-10-12:
A few remarks on the code:
* Please check if this runs properly if there is no vebus service on the D-Bus. I see no checks of the vebus service name against None, so systemcalc may crash.
* Why check the vebus main state before checking /Bms/AllowToDischarge and /BatteryOperationalLimits/MaxDischargeCurrent? The vebus state may change to 'off' if the BMS says discharge is not allowed. I think it is save to skip the check on the main state for those situations.
* Another reason for passthru may be that the Ac-In setpoint has not been set for 2 minutes (this should not happen unless ESS mode in settings is set to 'External Control'). So it may be interesting to add an 'external control' state or reason.

izak, 2017-10-19:
So after a few tries, which I didn't like, this is the basic idea.

1. The confusing part about the main display -- which shows the vebus state -- is that it appears to be charging (showing Bulk, Absorption or Float) while the battery is actually discharging. The simplest way to deal with this, is to simply add a state for discharging and show that instead.

2. We've been back and forth on whether Sustain is a flag or a state, but I'm leaning now towards making it a state.

3. In addition, when the system is not in the state you expect, we want to display some kind of reason why. These conditions might be present simultaneously, for example, you might have a condition where the BMS indicated that discharge is not allowed, while also having sustain active. So each flag will live on a separate DBUS path. The flags we want to indicate would be:

   * SOC is low
   * Battery life is active (this will go WITH low SoC)
   * Discharge disabled by BMS
   * Charge disabled by BMS
   * Slow Charge is active
   * Maximum inverter power is set to zero

izak, 2017-10-24:
Some feedback required. I'm not the world's most prolific UI designer, but this is what I came up with.

Normal tiled view, with three active conditions shown.
[image]

The balls-overview, with the conditions shown in a red-ish bubble.

[image]

izak, 2017-11-01:
Systemcalc now has unit tests as well. Branch is [here][1] for review.

[1]: https://github.com/victronenergy/dbus-systemcalc-py/commits/feature/systemstate

edevries-victron, 2017-11-01:
The states are implemented as a series of flags. How does that translate to the GUI? Some kind of priority system perhaps?

izak, 2017-11-02:
It prints out a #n on the screen for each condition. So in the picture above it is saying:

```
#1: SoC is low
#2: BatteryLife is active
#5: Slow Charge is in progress
```

Of course I faked that situation, you won't actually ever get a `#5` together with a `#1`.

The full error list for now:

```
#1: SoC is low (ESS state is 5 or 11)
#2: BatteryLife is active, this shows up with #1 if ess state is 5.
#3: Charging disabled by BMS (works for both vebus or can-bus)
#4: Discharge disabled by BMS
#5: Slow charge (ESS state is 6 or 12)
#6: Maximum inverter power is set to zero
```

The implementation on the gui side simply looks which flags are set and prints out the corresponding index. That's the simplest implementation for now, but it might change in future if we add or remove things.

edevries-victron, 2017-11-02:
Some remarks as result of my review:
* There's a problem with the UserLimited flag: in the comment above it is defined as 'Maximum inverter power is set to zero'). However, the source code also checks the MaxChargePower setting. I guess the source code is correct: we want to check for user limits on charge and discharge, just like we do with the BMS. For consistency, wouldn't it be better to introduce an extra flag. 
Like this: 
`#6: Charging disabled by user`
`#7: Discharge disabled by user`
There is another complication here: on system where BOL is enabled (vebus firmware >= 0x415), the MaxDischargePower setting (a.k.a. max inverter power) is no longer used. Instead the user can set the maximum charge current setting (`com.victronenergy.settings/Settings/SystemSetup/MaxChargeCurrent`). This setting can also be set to 0 in order to disable charging.

* are you sure the `Bms/AllowToDischarge` and `Bms/AllowToCharge` items in the vebus service always have valid values even if there is no VE.Bus BMS present? If there is no battery present with BOL parameters, the BOL parameters on the vebus service will be invalid. If the VE.Bus BMS parameters are invalid (or don't exist) as well the `bms_state` function will return `(None, None)` which will cause `ChargeDisabled` and `DischargeDisabled` to be `True`. I think it's better to handle this situation so that those flags are set to `False`, because there's nothing to stop the system from charge/discharge.

* there are some magic values in the code (vebus mainstate and ess state values). The code would be better readable if there were replaced with constants.

* you check against ESS state 12, which is not in used anymore, but you do not check against state 8, which is not used either (state 8 and 12 are the remnants of an earlier attempt to implement the recharge function).

izak, 2017-11-03:
* As discussed, we decided that the "Charging disabled by user" flag simply never shows up if you're not running BOL firmware.

* If `Bms/AllowToDischarge` or `Bms/AllowToCharge` doesn't exist, I'll end up with two `None`s. That will cause the branch on the next line to be taken and the existence of BOL parameters will be checked next. If those parameters also doesn't exist, it returns None, and since None is not equal to zero, it will return `(True, True)` which is what we want. If BOL parameters do exist, we return `False` only if they are zero. So I think this logic is okay. If there is a vebus BMS, then we might return integers instead of bools, so I decided to wrap them in `bool()` to force the type.

* Fixed the magic numbers and I handle states 8 and 12 the same now.

mpvader, 2017-11-08:
in v2.12~8

closed 2017-11-08 by mpvader

mpvader, 2017-11-29:
@izak what does this code for non-ESS systems? Nothing? (which is fine, I just want to make sure)

mkrenzke, 2017-11-29:
What I understood from @izak is that it passes through the vebus state in this case.

izak, 2017-11-29:
On a non-ESS setup it just passes through the VE.Bus state. Which is essentially "nothing".

mpvader, 2018-02-13:
@izak  : all of this is DVCC only, right?