victronenergy/venus #205: gpsd support

jwalck, 2017-10-05:
I have a ship with a Venus GX on board which I'd like to feed locationing, but I don't see a reason why it would need a dedicated GNSS receiver of its own. On board there is already another computer with a good receiver and installed antenna logging the ship location and feeding the location to a plotter computer over gpsd.

The simplest solution would be if Venus implemented gpsd configuration next to the dedicated USB GNSS configuration.

For background information, [gpsd](https://en.wikipedia.org/wiki/Gpsd) is a daemon that provides locationing data from one GNSS receiver to multiple consumers over a IP network.

An alternative approach would be if Venus would implement a gpsd server and I'd connect a receiver directly to the Venus. Personally I find the first option most compelling but optimally Venus would implement both.

mpvader, 2017-10-06:
hi @jwalck sounds like a nice addition. Will you implement this yourself?

jwalck, 2017-10-06:
I might, but it's not on the top of my stack right now. I imagine getting the functionality in place won't be hard but I haven't touched nor looked at the front-end (CCGX-style website).

mpvader, 2018-07-12:
hi @jwalck, I'm closing up stagnant issues, so closing this one too. Welcome though to pick it back up!

Regarding gui / front-end: I don't know if there needs to be any frontend? Is there configuration that is required for gpsd? Running a php website is not difficult.

closed 2018-07-12 by mpvader