victronenergy/venus #206: ESS: 'Phase Compensation' setting is confusing

edevries-victron, 2017-10-09:
It is unclear what the meaning of the 'Phase Compensation' setting is. The setting is only visible on multi phase systems. If it is enabled, not all phases will be regulated to 0W, only the total power. This will prevent moving power from one phase to the other via the DC system.

### Proposal
The 'Phase compensation' on/off switch is replaced with a MbItemOptions (similar to a combo-box).
The name of the combo-box is 'Phase balancing'. The options are:
* Total power will be regulated to 0W, but individual phases may take power from the grid or feedback power. In this mode the power flowing from one phase to another via the DC bus will be minimized.
* Balance all phases to 0W. This may cause additional energy loss, because extra power has to be sent over DC cables between Multis/Quattros. Chose this option only if your electricity provider does not allow regulating the total power.

The first option is the default.

@izak Do you think the use of the word 'regulate' here is correct? It sounds a bit strange to me, because I associate the word with laws (regulations), not with control loops.

There are 2 problems with the MbItemOptions control: it does not support multi line text in the sub-menu. 
It would also be nice to be able to show a short text describing the current choice when the options sub-menu is not open (like 'Balanced phases' or 'Unbalanced phases'), because the full text will not fit in a single line together with the description of the setting.

izak, 2017-10-10:
In electronics devices that ensure constant voltage/current are known as regulators and the function is known as regulation, so the term at least doesn't seem out of place.

The only other idea I had was to use the term counterbalance, since technically you're counterbalancing one phase with another to get equilibrium at zero across all three. I'm not sure that helps at all.

edevries-victron, 2017-10-10:
Maybe using 'Balanced' and 'Counterbalanced' as short description (shown in the ESS settings menu) is OK. If the setting is activated (that is, the MbItemOptions is opened to show the options), the long texts from my proposal are shown, which explain in some detail how it works.

closed 2020-03-13 by izak