victronenergy/venus #208: Move to a newer OE / Yocto version

mpvader, 2017-10-09:
Currently we are on Jethro (with all machines, hurray).

Info on all version [here](https://wiki.yoctoproject.org/wiki/Releases).

mansr, 2017-10-11:
There's a Pyro based build running on my Beaglebone.

mpvader, 2017-10-11:
That was quick. Fairly easy move then?

mansr, 2017-10-11:
It took some patching, but nothing major. Of course, I only said it runs, not that everything works, although I don't see anything obviously broken at a glance.

mansr, 2017-10-12:
Now running on ccgx as well.

mansr, 2017-10-17:
Summary of changes to Victron layers:

- Deleted recipes and patches from meta-alternatives and meta-third-party that are now redundant:
  - curl (patch)
  - daemontools
  - gdb (patch)
  - hiawatha (patch)
  - inotify-tools
  - libevent
  - php
  - python-lxml
  - python-requests
  - python-pycrypto
  - stress
  - tinymembench
  - tzdata
  - wpa_supplicant

- Added recipes/patches:
  - python-pygobject: various Victron scripts require version 2.x which is no longer in OE.
  - u-boot: backport patches to build with gcc-6.

- Fixed dependencies. Newer bitbake is stricter with dependencies, requiring every recipe to declare exactly what it needs. A few of our recipes had incomplete dependency lists.

- Update obsolete getVar/setVar usage.

- Use CFLAGS/LDFLAGS properly. New QA checks insist on certain flags being used. Some recipes calling $CC directly need updating.

Some changes are applicable already to the jethro build. These are on the [mans/misc](https://github.com/victronenergy/meta-victronenergy/commits/mans/misc) branch.

jhofstee, 2018-02-16:
(Almost) all of this got merged in master a while. The remainder is in https://github.com/victronenergy/meta-victronenergy/commits/mans/misc2. I guess we are better of testing connmand again after switching to a newer OE version. https://github.com/victronenergy/meta-victronenergy/commits/mans/misc can be deleted.

jhofstee, 2018-03-01:
included in v2.20~6

closed 2018-03-01 by jhofstee

mansr, 2018-03-10:
Things to test, in no particular order:
- installer image
- sd/usb auto-mount, vrm logger
- ethernet
- wifi
- access point
- swupdate: online, offline
- backup image
- production test
- external interfaces
  - CAN: profiles
  - VE.Bus
  - relays
  - digital inputs
  - analogue inputs
  - VE.Direct
- LEDs
- GPS
- serial-starter
- GUI
- daemontools services
- VRM logger
- remote console
  - LAN
  - VRM
- crond: automatic update
- disk space
- bootlog/dmesg

ThiemoVanEngelen, 2018-03-19:
Wifi dongles that are officially supported:
- part number BPP900100100 - Zyxel NWD2105 - USB 0586:341e - RT chipset 3070, rev 0201, RF chipset 0005
- part number BPP900100200 - Nano USB - USB 148f:5370 - RT chipset 5390, rev 0502, RF chipset 5370
- part number BPP900200100 - Startech USB300WN2X2D - USB 148f:3572 - RT chipset 3572, rev 0223, RF chipset 0009
- part number BPP900200200 - Gembird WNP-UA-002 - USB 148f:3072 - RT chipset 3071, rev 021c, RF chipset 0008

mpvader, 2018-03-20:
test list moved to https://github.com/victronenergy/venus-private/wiki/Things-to-test