victronenergy/venus #209: Tank volume is limited to 1000 liters, 219 gallons or 264 US gallons

izak, 2017-10-16:
On the Venus GX, you cannot set tank volume larger than 1000 liters or the associated conversions in other units.

izak, 2017-10-16:
The tank capacity is limited to one cubic meter in [code][1]. It should be easy enough to change it, but first I'd need to know if there was a specific reason for choosing 1 cubic meter.

[1]: https://github.com/victronenergy/dbus-adc/blob/master/software/inc/sensors.h#L51

mpvader, 2017-10-18:
@izak Lets increase the max to some huge number

izak, 2017-10-18:
The NMEA2000 packet has 4 bytes for this, with a resolution of 0.1, which pegs the maximum at a little over 400 million liter. A small tanker starts at 10 000 DWT, which is ten million liters assuming it is water. I think we should be fine setting it to 1 million liters, or 1000 cubic meter.

mpvader, 2017-10-18:
Ok

izak, 2017-10-20:
Just bumping up the maximum seems to work perfectly. I'll make a pull request for inclusion into the main release.

[image]

mpvader, 2017-10-20:
You've full access now, please make the release of dbus-adc yourself. See previous tags for how it's done

mpvader, 2017-10-27:
In v2.12~5

closed 2017-10-27 by mpvader