victronenergy/venus #211: dbus-vecan: add support for 24 bit version numbers

mpvader, 2017-10-19:
See [vregs.xml in velib](https://github.com/victronenergy/velib/blob/master/data/protocols/vregs.xml#L2126) for the new definition.

```
dbus -y com.victronenergy.solarcharger.socketcan_can0_di0_uc16252 /FirmwareVersion GetText
value = 'v2.05'
```

martinbosma, 2017-10-30:
This is added here: https://github.com/victronenergy/dbus-vecan/commit/37ffa076fbce8b56567d9677e4372624841fe375.

Version 0x011000 (old style v1.10.00) is dispalyed as follows:
```
root@ccgx:~# dbus -y com.victronenergy.battery.socketcan_can0_di8_uc0 /FirmwareVersion GetValue
value = 69632
root@ccgx:~# dbus -y com.victronenergy.battery.socketcan_can0_di8_uc0 /FirmwareVersion GetText
value = 'v1.10-beta-00'
```
Version 0x0110FF (old style v1.10.FF) is displayed as follows:
```
root@ccgx:~# dbus -y com.victronenergy.battery.socketcan_can0_di8_uc0 /FirmwareVersion GetValue
value = 69887
root@ccgx:~# dbus -y com.victronenergy.battery.socketcan_can0_di8_uc0 /FirmwareVersion GetText
value = 'v1.10'
```
And finally 0x0110 is still displayed as:
```
root@ccgx:~# dbus -y com.victronenergy.battery.socketcan_can0_di8_uc0 /FirmwareVersion GetValue
value = 272
root@ccgx:~# dbus -y com.victronenergy.battery.socketcan_can0_di8_uc0 /FirmwareVersion GetText
value = 'v1.10'
```

@mpvader Can you closed this, cause apparently I can't.

mpvader, 2017-10-30:
> @mpvader Can you close this, cause apparently I can't.

Please first make a release of dbus-vecan, and add that to the Venus todo list, make sure to include a link to this issue on the todolist.

Also: put a comment in this issue, mentioning the version of dbus-vecan that closes it.

Then still leave it open: Jeroen or myself will close the issue once its in Venus master.

Like here for example:
https://github.com/victronenergy/venus/issues/151

martinbosma, 2017-10-30:
Added in dbus-vecan v2.23

mpvader, 2017-11-03:
in v2.12~6

closed 2017-11-03 by mpvader