victronenergy/venus #212: vedirect-interface: add support for 24 bit version numbers

mpvader, 2017-10-19:
See [vregs.xml in velib](https://github.com/victronenergy/velib/blob/master/data/protocols/vregs.xml#L2126) for the new definition.

```
# dbus -y com.victronenergy.battery.ttyO2 /FirmwareVersion GetText
value = 'v3.08'
```

mpvader, 2018-07-12:
Thiemo warned about some tricky bits:
1. Are we sure that every device gives a correct answer for VE_REG_APP_VERSION?
2. What do we do with the BMV60x where we cannot send hex? It is not really a problem because there is no 3 byte version inside the BMV, but we do need conversion from the 2 byte text frame version.
3. Do we need to ignore the version in the text frame for devices that cannot talk hex? I am not sure if the ve.direct stack in vedirect_interface has knowledge if hex trafic is allowed.

izak, 2018-11-14:
A new text item `FWE` was added that contains the 24-bit version number. This answers all the tricky questions above.

Implementation [here][1] and [here][2].

[1]: https://git.victronenergy.com/victron/vedirect_consumer/tree/un24-versions
[2]: https://github.com/victronenergy/vedirect_interface/commits/un24-versions

mpvader, 2018-12-03:
Izak is done and is waiting for review or inclusion

izak, 2019-01-28:
Included in Venus v2.30~24.

closed 2019-01-28 by izak