victronenergy/venus #213: issues during installation

Zachmoore111, 2017-10-23:
Hey Guys,

I'm having a few issues with getting VRM logger running, I've been following the instructions listed here https://github.com/victronenergy/venus/wiki/raspberrypi-install-venus-packages, but i've run into a few problems and i don't know what to do

My setup:
Raspberry pi 3
Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
Installed with wlan0 but I've also had issues with the huawei E3372 LTE usb stick i plan on running it with
2x Victron ve-direct usb cables

Problem
root@raspberrypi:~# ls /etc/service/
dbus-fronius  dbus-qwacs  dbus-systemcalc-py  dbus-vebus-to-pvinverter	localsettings  vrmlogger

I'm missing all the files to allow me to configure the vedirect interface, and I'm unable to register the installation on the VRM website. I have tried the MAC address from both Eth0 and wlan0

I have a log of the terminal session from start of installation till checked to see if /etc/service/vedirect-interface/run existed, happy to submit that if it would be useful

I'm pretty lost with all this, I've had a little bit of help from a friend who is a sysadmin, to see if i'm doing anything wrong with my commands, missing something, but he hasn't been able to help much with his limited free time

mpvader, 2017-10-23:
Hi @Zachmoore111, please use the dev-venus mailinglist for this. The packages are not something we support as a company. I’m relying on support from other users to help you.

To make it easier for yourself, I recommend using the image instead of the packages. Also for the inage there is only community support. But the advantage is that Its easier to get up and running.

Success! I know from experience that it can be a bit daunting when new it at all.

closed 2017-10-23 by mpvader

mpvader, 2017-10-23:
ps, @izak updated the instructions today, see here for the new version:

https://github.com/victronenergy/venus/wiki/raspberrypi-install-venus-packages