victronenergy/venus #214: qt & qtquick versions

mpvader, 2017-10-24:
This discussion flares up regularly, below list is to get the facts in a row. Welcome to comment if you have more to say about it.

## Currently used versions
* Qt 4.8.7 (incl built-in VNC with multiple screens: QWS)
* QtQuick 1.1

Note that, besides the GUI, we have various other applications that use the qt libraries. Not for display, but for string handling etcetera.

## Notes about new version:

#### QtQuick 2.*
* Adds graphics acceleration (OpenGL ES).
* Until Qt 5.8 (or something) graphics acceleration was required. After that a software fallback mode was implemented. With a drawback: QtQuick 2.* with software fallback is reportedly slower than a QtQuick 1.1 that doesn't use graphics implementation in the first place.
* QtQuick 2.* is the version we use, or will use soon, in VictronConnect. Obvious advantage of using the same in Venus.

#### Qt 5.*
* In Qt 5.8 support for QtQuick 1 was dropped. OpenEmbedded still supports it though.

## VNC
- QWS (which has the built-in VNC we use) was dropped in 5.0.
- In Qt 5.8 there is some new support for VNC again. http://www.jumpnowtek.com/rpi/Running-Qt5-Apps-Over-VNC.html. We expect that, unlike on 4.8.7, its not possible to output the gui to both VNC and a display (such as the CCGX) at the same time.
- Qt5 does support Wayland. And Wayland is also working on VNC: https://www.ctrl.blog/entry/wayland-gnome-remote-desktop.

Besides above, we could also run X-server, which we currently don't.

## Summary
So, to summarize, in order to move to Qt5 and to the new QtQuick version, there are two issues to tackle:
#### 1. Graphics acceleration
either test how well QtQuick 2 works with the fallback, or look into making the graphics accelaration work on all boards running Venus:
* CCGX: has graphics acceleration, but is not supported in mainline kernels (which we currently use in Venus)
* BeagleBone: same as CCGX.
* CANvu500: has graphics acceleration, and its supported in the mainline kernel we use (right @mansr?)
* NanoPi: ??

#### 2. VNC

## Why update?

Do we need to update? No we don't. These are the arguments to do so:
1) On the CANvu500 we do want to offer Qt5 and also the use of graphics acceleration. Not perse for the Venus GUI, but for other GUIs.
2) Then there is the advantage for Jesus, working on both VictronConnect and the CCGX
3) QtQuick 2 adds Canvas, which will be very helpful in the overviews

Last one would be the dangerous argument that newer is better.

And next last one could be bluetooth.

## Various notes
- Shipping two versions of the Qt libraries at the same time is possible, but increase memory usage (RAM), as well as non-volatile memory.
- Running two GUIs at the same time (on the CCGX, one for the display and one for VNC) is not advisable either, besides memory there will also be a CPU issue.

mansr, 2017-10-27:
Yes, the canvu500 has working graphics acceleration. The ccgx screen is so small that I doubt the performance drop with the software rendering in QtQuick will be of any concern.

mpvader, 2017-11-08:
In v2.12~6, a relevant change was made:

```
v2.12~6 - develop testing
- Qt 5.9 is now available in the SDK and on the CANvu500
  - Qt 5.9 can now use Qt Quick 2 on eglfs and linuxfb
  - Only the canvu500 has support for egl. A demo is included, to run use:
     svc -d /service/gui && Qt5_CinematicExperience
```

mpvader, 2017-12-12:
for the record: @jhofstee has been experimenting with using vnc from X11, but then without X11.

`x11vnc -noipv6 -rawfb /dev/fb0 -pipeinput UINPUT:direct_key=/dev/input/event0,nouinput -shared -forever`

Possible advantages:
- multiple clients at the same time
- (probably) having lower cpu cost
- this opens the route to using a newer Qt / QtQuick version, including vnc-ing a display that has been drawed using hardware accelaration

first trials work, to be continued.

mpvader, 2018-08-21:
VNC issue: 
https://github.com/victronenergy/venus/issues/292

mpvader, 2019-01-25:
An update after discussing with Thiemo and Jeroen today:

A month ago; Thiemo tested running VictronConnect on a CANvu 500; and using VictronConnect as a base for a new Venus GUI was discussed with Jesus, Thiemo and Reaktor. The conclusions where that (a) VictronConnect is slow; bug can probably be made faster. And (b) using all of VictronConnect is not a good idea (landscape layout vs portrait; and difference in architecture: D-Bus vs VE.Interface) ; but re-using or sharing its components and design for Device pages does make sense.

In general, moving from QtQuick 1.1 and Qt 4.8 to QtQuick 2 and Qt 5 is still wanted. The main issues are:

- VNC: solvable.
- CCGX doesn't have hardware acceleration. So either we use the Qt software renderer (which just skips all hardware accelerated operations), or some other software renderer such as MESA -if that supports OpenEGLS2; and live with whatever disadvantages that has. Or, CCGX keeps the old GUI, and we only maintain that for new settings, and let it be.

Steps to move forward are first to (a) study the [Qt Quick 2D Renderer documentation incl. limitations](http://doc.qt.io/QtQuick2DRenderer/), and (b) check how bad the Qt Quick 2D Renderer is, both in looks and performance. A simple way to check the looks can (probably) be done with VictronConnect: compile it for use with that rendered and see what happens.

closed 2025-01-31 by mpvader