victronenergy/venus #215: gui: don't pretend that there is a VE.Bus BMS

mpvader, 2017-10-26:
On a system with no VE.Bus BMS, and also no VE.Bus BMS installed, the gui shows this:
[image]

instead, I'd prefer an empty menu, with just a single line 'No VE.Bus BMS installed'.

The dbus info:
```
Bms/AllowToCharge                       1
Bms/AllowToChargeRate                   0
Bms/AllowToDischarge                    1
Bms/BmsExpected                         0
Bms/Error                               0
```

I noticed this on the BMZ, and also other, test systems in Groningen.

mpvader, 2017-10-26:
@vejpasop can we simply put in that line when `Bms/BmsExpected == 0` ?

Or when `Bms/BmsExpected ==0 && Bms/Error == 0` ?

By the way, what are the definitions of Bms/Error ?

jhofstee, 2017-10-26:
@mpvader 
" can we simply put in that line when /Bms/BmsExpected == 0 "
yes, there is no valid BmsError flag if there is no BmsExpected. Both are found in the same assistant unless things changed. Up to @vejpasop  to confirm.

"Or when /Bms/BmsExpected ==0 && /Bms/Error == 0"
That would be inconsistent with the rest of the UI.

"By the way, what are the definitions of /Bms/Error"
It is a boolean.

I assigned it to myself, since it will be two lines of code.

vejpasop, 2017-10-27:
" can we simply put in that line when /Bms/BmsExpected == 0"
Yes.. Confirmed.

vejpasop, 2017-11-17:
For this to work with the new ESS assistant "two signal" BMS option, mk2-dbus v2.59 is required.
With previous versions  of mk2-dbus /Bms/BmsExpected will remain 0 when the "two signal" BMS is configured.

closed 2017-12-05 by jhofstee