victronenergy/venus #216: dbus-modbustcp improvements

edevries-victron, 2017-11-01:
Some dbus-modbustcp improvements I have been working on in the past year, which have not been released yet:
* Update source to use VeQItems
* Publish a D-Bus (com.victronenergy.modbus) service which supplies the unit ID for each D-Bus service. Many customer questions about modbus TCP come down to misunderstanding how unit ID's are tied to D-Bus services. The service also publishes the last error (if any).  There is also a gui counterpart: it works, but does not look really good.

Todo's:
- [ ] Push the WIP branch to github
- [ ] Do some more testing (especially duration test: potential memory leaks & performance)

3dGrabber, 2018-05-16:
#### gui - WIP_modbustcp_diagnostics branch

- changes are only in view/qml, no code changes
- adds `PageSettingsModbusTcp.qml` and subpages to `PageSettings.qml`
- gui is functional, tested with demo mode and on a client’s machine
- rebase yielded merge conflict: resolved by removing `Modbus/TCP` menu entry from `PageSettingsServices.qml` (moved to `PageSettingsModbusTcp.qml`)
- tested, working in conjunction with the `WIP_qitems_and_diagnostics_service` branch of `dbus_modbustcp`, see below

#### dbus_modbustcp - WIP_qitems_and_diagnostics_service branch

- many commits, many changes, in many files
- changes seem reasonable, but I can’t pretend I understand 100% of them
- changes could be rebased without conflicts
- compiles without errors and warnings
- tested on test and a clients machine (ccgx), sending various modbus requests, target replied with the expected values
- continuously polled the machine for more than an hour, memory footprint did not increase => memory leak unlikely
- works in conjunction with the `WIP_modbustcp_diagnostics` gui, see above
- the branch adds some unit tests (`/test/main.cpp`): those tests won’t readily run on a target. needs `libQtTestE.so` which can be installed using `opkg install libqtteste4`
unit tests were run locally on x86_64, all of them passed.
- the branch adds some integration tests with `modbustcp_tb.py`. those need *a vebus device [..] and a BMV to be connected to the CCGX* all tests passed, except `test_6`
(because of missing BMV, I do not have access to a system with a BMV)

mpvader, 2018-07-12:
hey @3dGrabber, I'm looking for your patches in the [gui branch](https://github.com/victronenergy/gui/commits/WIP_modbustcp_diagnostics), but can't find any?

mpvader, 2019-06-22:
Listed on the v2.40 todo, for Izak.

closed 2019-08-15 by mpvader