victronenergy/venus #217: Read-only rootfs

mansr, 2017-11-06:
## Advantages
### Delta updates
A read-only rootfs enables delta updates where only the difference between versions needs to be downloaded. On slow, expensive, or unreliable links, this is a great advantage.

### Robustness
As each boot is identical, system behaviour becomes more predictable. We have already had an issue that was made more difficult to debug due to the rootfs changing between the first and subsequent boots.

The risk of filesystem corruption on power failure or similar is also virtually eliminated. Although ext4 and ubifs are robust at the filesystem level, applications might still be confused in case of a partially written file or other unexpected state.

## Implementation
With the switch to image updates, most modifications to the rootfs have already been removed. A few, however, remain:
- Auto-created mount points in /media. This is fixed by making /media a symlink to a volatile location.
- The kernel random number generator state is saved across reboots. The file used for this can easily be moved to the persistent /data partition.
- Daemontools creates various control files under the service directories. As these do not need to persist across reboots, overlayfs with a volatile upper layer handles these.
- Files in /etc:
  - simple-upnpd.xml: Created on (first) boot from a template. No need for persistence as creation is quick.
  - shadow: If the user configures a root password in the gui, the shadow file is updated. Currently this survives a reboot but not an upgrade. As such, making the change volatile over any reboot is probably acceptable.

### Post-install scripts
OE packages can include scripts to be run after installation. For most packages, this can be done on the build host when creating the image. A few, however, do not support this, and they are deferred to the first boot, after which they are deleted. Obviously, this entails modifying the rootfs (even if the script itself does not), and this thus not compatible with a read-only configuration.

The only affected package currently in Venus is localsettings. This can be handled either by modifying the package to not use the postinst functionality or by devising a different solution for running scripts on an upgrade.

### Compiled Python files
Python creates compiled files (.pyc) the first time a module is loaded. While optional, the presence of these files reduces startup time. The Victron packages using Python should thus be modified to pre-compile the modules during image creation.

### Daemontools service state
For a few services, the gui alters the default state (by creating or deleting the `down` file) in response to configuration changes. These changes are expected to persist across reboots. Overlayfs with a non-volatile upper layer would permit this, but it would also needlessly store the control files created by daemontools (see above). An alternative mechanism for managing the initial state of these services would thus be preferable.

### Optional packages
Installing additional packages is impossible unless first remounting the rootfs read/write. Doing so will obviously prevent delta updates. On devices using eMMC storage, this is otherwise not a problem. However, in order to use delta updates with ubifs, it is necessary to use static volumes which cannot be mounted read/write. On such devices, delta updates and the ability to install packages are thus mutually exclusive.

mansr, 2017-11-06:
## To do
- [ ] Integrate localsettings conversion in main script
- [ ] Make list of services managed by the gui
- [ ] Devise new scheme for managing said services
- [ ] Compile Python modules offline

mansr, 2017-11-06:
## Services managed by the gui
Service | Settings
--------|---------
vegetty | Services/Console
dbus-modbustcp | Services/Modbus
mosquitto | Services/Mqtt
dbus-mqtt | Services/Mqtt
openssh | System/RemoteSupport
dbus-pump | Relay/Function
ssh-tunnel | System/RemoteSupport, System/VncInternet
vrmpubnub | Services/Vrmpubnub
websockify-c | System/VncLocal
dbus-generator-starter | Relay/Function, Services/FischerPandaAutoStartStop
hostapd | Services/AccessPoint
vecan-dbus.* | Canbus/*/Profile
can-bus-bms.* | Canbus/*/Profile
dbus-valence.* | Canbus/*/Profile
dbus-motordrive.* | Canbus/*/Profile

mpvader, 2018-07-12:
https://github.com/victronenergy/meta-victronenergy/tree/mans/read-only-rootfs

Status: first we'll release v2.20. And then we'll look at canbus (application level), and thereafter this one is up for inclusion.

closed 2021-07-21 by jhofstee