victronenergy/venus #219: canvu500 gui issues

mansr, 2017-11-14:
Buttons appearing at the bottom of the screen don't work properly. Some settings use the check/cross buttons to accept or discard an entered value. However, tapping that area of the screen always gives the Pages or Menu action. These settings are thus impossible to change.

mkrenzke, 2018-09-17:
There is no way to change the settings where there is no selection behind but some kind of free field to input a value. With the buttons of a CCGX you can roll the numbers/characters by pressing up and down. Without buttons you can't.

Where I found this issue:
- Settings -> General -> Set root password
- Settings -> System setup -> User defined name
- Settings -> Date & Time -> Date/Time local
- Settings -> Remote Console -> Enable password check
- Settings -> Wifi -> Password
- Settings -> Ethernet -> IP configuration -> Manual
...there might be more

Some of them (e.g. Set root password) show an information "how to input" below the field when you select it. Wouldn't it be a solution to show an on-screen-keyboard there instead for touch devices? There could come up some more in the future.

jhofstee, 2019-03-20:
there is a touch keyboard nowadays allowing to edit these field with touch only, so closing this.

closed 2019-03-20 by jhofstee