victronenergy/venus #225: Investigate what happens if we distribute vebus voltage to charge controllers

izak, 2017-11-22:
# Problem description

Voltages might be at slightly different levels throughout an installation because of voltage drops over cable runs. This is unavoidable, as even a 1% drop at full load is considered good, but could amount to a not-insignificant difference in voltages. The ideal is that all equipment in an ESS system agrees on the battery voltage.

# Suggested solution

The suggested solution is to copy the vebus voltage to the solar chargers, so that everyone agrees on the voltage. Wiring voltage sense to your multi also makes that reading accurate.

# What could happen when there is no v-sense installed

The VEBUS voltage is potentially going to be inaccurate, depending on the amount of current being drawn and the direction of flow.

## When the multi is charging

While charging, the vebus voltage will be higher than the actual battery voltage. This voltage will be copied to the solar chargers and they too will think the voltage is higher than it really is.

Possible side effects:
 * The solar charger could back off on charging when it sees the higher distributed voltage, where in the past it would continue based on its own lower measurement. This could potentially undercharge, or at least hamper absorption. For example, on a lead acid bank, this could collectively set solar chargers to absorb at 0.5V lower than required.

## When the multi is discharging

While discharging, the vebus voltage will be lower than the actual battery voltage. Distributing this value will cause solar chargers to charge the batteries to a higher actual voltage.

Possible side effects:
 * The solar charge controllers will think there is room to raise the voltage some more, and will essentially over-voltage the batteries by about the same amount as the voltage drop on the cable to the Multi. Again, using the example of a lead-acid battery designed to absorb at 14.8V, this could push the voltage to 15V.
* Temperature compensation. I'm not sure exactly how this works, but if temperature compensation brings the voltage down a bit because of ambient temperature, but you use the wrong voltage, you nullify the effect of temperature compensation.

edevries-victron, 2017-11-23:
A first implentation was released in systemcalc v1.44-bol6:
* The implementation uses `com.victronenergy.system/Dc/Battery/Voltage`.
* A new D-Bus path `/Dc/Battery/VoltageService` was introduced to keep track of the service from which the voltage was taken, so that we can prevent to copy a voltage to the source service.
* If the vebus does not support setting a voltage (only firmware 420beta can do this), the voltage is not copied to the vebus, but still copied to the solar chargers.

mpvader, 2018-07-12:
Closing: shared voltage sense was a bad idea.

closed 2018-07-12 by mpvader