victronenergy/venus #226: change bitrate of slcan on nanopi port not supported with ip command

fhuebner, 2017-11-23:
On the nanopi port we use a slcan interface. 

it turns out that it is currently not possible to set the bitrate with the ip command.
`ip link set up can0 type can bitrate 250000`
`RTNETLINK answers: Operation not supported`

so currently we will use a workaround to get the slcan device run at the desired bitrate:
1. get triggered by dbus on profile change
2. kill the currently running slcand 
3. start a new instance with bitrate selected as follows:
- Profile = 0 -> disable
- Profile = 3 -> 500kBit/s
- all other -> 250kBit/s

(to replace it easily later on: we will create a seperate service for that)

mpvader, 2017-11-23:
hi @fhuebner, why not patch the gui instead of adding a service? What we could do is add the patch in OE, machine specific for the nanopi, instead of in the gui repo.

mkrenzke, 2017-11-27:
@mpvader If the gui is the only place where can profile changes get handled, this seems to be a viable way.

fhuebner, 2017-11-27:
@mpvader did it as patch for the gui in OE. thanks for the hint!

mpvader, 2017-12-22:
@fhuebner this was fixed, right? then we can close this issue

mpvader, 2018-02-23:
@mansr will fix this in a few weeks, as part of his work on making Venus run on the new display module using the nanopi neo core.