victronenergy/venus #227: Venus limited export target

Wholistic, 2017-11-26:
Victron - Venus limited export

A feature is required to set target upper limit for grid export.

Multi, power meter, Mppt and CCGX are presently constrained to a binary ‘feed excess solar charger power back’ as on or off.

A soft target limit, with the same variability/flexibility as the current grid set point (at 0), but works instead as a upper boundary as a programmable value of 0-30000 W.

As the Venus cannot control all the inputs ( such as external AC Pv on an energy meter ) this will only be a guide point for the control of the mppt. 

Strict export limitation may not be possible with the current ESS implementation, but a sufficiently low target should be acceptable ( eg 2000W target export limit, on the 3000W supplier export limit with a 5000W solar array).

Batteries will charge with the full 5kW. When the batteries are full, Mppt will ramp down to 2kW, as clouds and loads change, the actual export value will fluctuate as best as the equipment is able similar to the 0 point hunt. I have personally seen this as high as 600W out, but is usually corrected in seconds.

kristofgoethals, 2018-01-09:
i agree on this issue...since you need to "setup" the max. grid load (current)
"must" be possible to limite the export with that value?

like me, i will have 200A [when max. sun] (panel size on DC via victron mppt) on solar, so when full and only used 20A on load...that will blow my fuse on export?

Wholistic, 2018-01-10:
Kris, 
You will still be limited by primarily the MPPT and then the Inverter power of your Multi. 

Your fuses should be sized first to protect your wires, and second to be large enough so that the output of the MPPT and Multi is not able to overload the fuse.

This feature request is more for the case when there is multiple solar supplies, which are visible to the CCGX via a grid power meter (eg Guzzi). 
In normal ESS zero export functionality, the CCGX will control the Multiplus and tell it to push power out to run loads that are not necessarily connected to the AC output of the multi (eg a non-essential oven load). 
The current alternative option is that all solar is pushed out to the grid, regardless of if it is required by loads or not. Surplus power to load is exported.

This can cause issues where wire sizes on the grid supply AC are too small or too long, and AC grid voltage will rise to unacceptable levels. 

If an export target limit feature is added, some measure of export is allowed, but not so much that it causes grid issues. This is usually prescribed by the grid distributor authority.

Wholistic, 2020-02-03:
implementation here - https://github.com/victronenergy/venus/issues/594

closed 2020-02-03 by Wholistic