victronenergy/venus #228: gpio: change & use symlinks created by gpio-export

mpvader, 2017-11-29:
- [ ] change gpio export to make symlinks to gpio instead of gpio/value
- [ ] change systemcalc to access & enumerate relays via gpio-export symlinks
- [ ] drop VE_RELAYS from the machine.conf files
- [ ] update digital inputs, it needs to add /value itself

mpvader, 2017-11-29:
the systemcalc change needs to be done in two branches, bol & master ..

mpvader, 2017-11-29:
gpio_pins.sh change:
https://github.com/victronenergy/meta-victronenergy/commit/6ea89fc2cad4d80ad53137785c21182c6070d4af

mansr, 2017-11-30:
mkx_pins.sh done:
victronenergy/meta-victronenergy@57a1378ef7e2c8bf9851a47d497f2fb7c1ac9e7f

izak, 2017-12-12:
digitalinputs done:
izak/dbus-digitalinputs@79826a1763dd0aa25bba3baa2d5cbc406f15b81a

izak, 2017-12-13:
dbus-systemcalc-py done:
victronenergy/dbus-systemcalc-py@3ab4170da390c3bb5d7f7aa8e5f8d307bad9f056

dbus-systemcalc-py-bol done:
victronenergy/dbus-systemcalc-py@2a7ec7a8115eaf6783432303c5ca04fc174951ad

closed 2018-03-02 by mpvader