victronenergy/venus #229: BYD firmware version wrong on vrm.

mpvader, 2017-11-30:
The same canbus data (`can0  35F   [8]  4C 69 04 0F 32 00 11 17`) shows as hex on vrm: v4.0F. While in the Ccgx it is shown in decimals.

Most probably this needs fixing in the Venus OS, fix is expected in the next v2.12~ test version.
[image]
[image]

mpvader, 2017-12-03:
The gui shows it correctly because it does a gettext.

BYD battery:
```
dbus -y com.victronenergy.battery.socketcan_can0 /FirmwareVersion GetValue
value = 1039   (= 0x40F => v4.0F in Victron notation, but BYD expects v4.15)
dbus -y com.victronenergy.battery.socketcan_can0 /FirmwareVersion GetText 
value = 'v4.15'
```

BMV:
```
# dbus -y com.victronenergy.battery.ttyO2 /FirmwareVersion GetValue
value = 776  (= 0x308 => v3.08 in Victron notation)
root@ccgx:~# dbus -y com.victronenergy.battery.ttyO2 /FirmwareVersion GetText 
value = 'v3.08'
```

One solution is just to no longer do the conversion from number to string in vrmlogger, but use the GetText for this path instead. (in the vrm database we store firmware values as formatted string).

@jhofstee any thoughts?

mpvader, 2017-12-03:
Conversion by vrmlogger is done here:

https://git.victronenergy.com/ccgx/dbus_vrm/blob/master/conversions.py#L23

and here:

https://github.com/victronenergy/velib_python/blob/master/ve_utils.py#L55

mpvader, 2017-12-04:
As discussed with @jhofstee, indeed the proper fix is to use the GetText value, and not the GetValue.

Which maybe means adding GetText values to dbus-monitor, and renaming functions like [these](https://github.com/victronenergy/velib_python/blob/master/dbusmonitor.py#L200) to handle_properties_changed().

izak, 2018-01-19:
Solution we decided on is to use a kind of HED (hex-coded decimal, coined from the similar term BCD).

Afterwards: 
```
root@ccgx:/opt/victronenergy/can-bus-bms# dbus -y com.victronenergy.battery.socketcan_can0 /FirmwareVersion GetText 
value = 'v4.15'
root@ccgx:/opt/victronenergy/can-bus-bms# dbus -y com.victronenergy.battery.socketcan_can0 /FirmwareVersion GetValue
value = 1045
```

1045 translates to 4.21 in decimal, which translates to 4.15 when rendered in hex.

mpvader, 2018-01-24:
Fixed in vrmlogger 2.90 & 2.91. waiting for inclusion in Venus

mpvader, 2018-02-13:
This was in v2.12 for a while, but then reverted because of issues. Is now scheduled for v2.13

closed 2018-02-26 by jhofstee