victronenergy/venus #233: systemcalc: fix PV DC-Coupled Power for MPPTs with load output

mpvader, 2017-12-08:
The load output state and current is being sent to the VRM Portal already, using dataAttributeCodes `SLs` and `SLI`. See [here](https://git.victronenergy.com/ccgx/dbus_vrm/blob/master/datalist.py#L218).

But, looks like systemcalc takes the batterypower (which can be negative it night when there is a load output..) instead of PV Power:

[image]

Community has also worked on this:
https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/victron-dev-venus/TtcogHiZF-Y/PPmTolJDBQAJ

mpvader, 2017-12-08:
A quick glance a system-calc [shows](https://github.com/victronenergy/dbus-systemcalc-py/blob/master/dbus_systemcalc.py#L41) that it only looks at battery current. The path that it should look at instead can be found on the [dbus page](https://github.com/victronenergy/venus/wiki/dbus#solar-chargers). And looks like we need /Yield/Power.

mpvader, 2019-10-05:
Re-opening, looking at this again it needs a proper test with a real unit before I'm sure that this works well. Tests:

- PV Available, current flowing from PV to both Battery & Load output terminals. Expected result:
  - Total PV current & power reported properly on UI in solar box.
  - Current/Power on load output reported properly in UI in DC System box.
  - Current/Power to battery reported properly in battery box.

- PV Available, but not enough to power both load and battery
  - Total PV current & power reported properly on UI in solar box.
  - Current/Power on load output reported properly in UI in DC System box.
  - Current/Power from the battery reported properly in the battery box.

- No PV Available, current flowing from battery to Load output. Expected result:
  - no solar yield reported (0W in the solar box)
  - but show current/power on load output in the DC System box on the GUI.
  - show the same current negatively on the battery box.

Test both with a BMV in place and configured, and without a BMV configured (if its installed or not doesn't matter ofcourse).

mpvader, 2019-10-05:
What I'm suspicious of is that it adds the current on the battery terminal & on the load output together to get solar yield. But that can never work well at night.

Here you can see that it doesn’t look at the yield as reported by the MPPT (its not doing anything with /Pv/V, /Pv/I nor /Yield/Power.
https://github.com/victronenergy/dbus-systemcalc-py/blob/master/dbus_systemcalc.py#L36

Function where it does all the calculations is here:
https://github.com/victronenergy/dbus-systemcalc-py/blob/master/dbus_systemcalc.py#L402

brendanwallison, 2023-05-05:
Hello, was this issue ever resolved? I am looking at a use case of a Raspberry Pi hooked up to the load terminal of a Victron MPPT charge controller. We are particularly interesting in gaining access to the logging features by installing Venus packages on Raspbian. But I am concerned that there are lingering issues with Venus properly fetching and using data from the MPPT.