victronenergy/venus #236: /data/venus informational tree is missing on Raspberry Pi

izak, 2017-12-13:
On the other machines we store information in files, for example `venus/installer-version` stores the version in use when this system was installed.

We should add `/data/venus/image-version` for the Raspberry Pi and similar architectures.

- [ ] Add `/data/venus/image-version` for the Raspberry Pi (izak)
- [ ] Add `/data/venus/image-version` for the NanoPi (felix)

izak, 2017-12-14:
I made this change for the Rpi in victronenergy/meta-victronenergy@3d17b541241fa2a41156fb203279b9db0fd0d4e1. It can possibly be used almost as-is for the nanopi.

mpvader, 2018-02-19:
@mansr this no longer applies to nanopi, right?

mansr, 2018-02-19:
No, the new version with eMMC doesn't need this.

closed 2018-02-19 by mpvader