victronenergy/venus #238: mqtt-n2k: enable service mess with devices addresses on Nmea 2000

Miliox, 2017-12-19:
The address of the devices on Nmea 2000 change constantly when starting mqtt-n2k service. Apparently, the address of the devices is stolen periodically and this force the address to change. This breaks the connection between devices on Nmea 2000.

My guess is that problem is caused by mishandling of the address conflict by mqtt-n2k. I am investigating the problem, but for now is better to keep the service disabled until a solution is found.

mqtt-n2k version: 0.1.3 on firmware v2.12~22 (20171217100631).

Miliox, 2018-01-03:
After some debugging and help from Eike, We found that it was caused by using the same address settings from MainInterface.

Changing to its own setting prevent the problem. This problem may happen again if two instances of mqtt-n2k execute on the same can interface.

mpvader, 2018-01-03:
hi @Miliox ok so you make a commit in mqtt-n2k that changes the code to use its own setting, which closes this issue?

@jhofstee let us know if there is any issue with this.

Regarding:
> This problem may happen again if two instances of mqtt-n2k execute on the same can interface. But that should not happen when running using service.

indeed, thats not a real problem.

Miliox, 2018-01-03:
Hi @mpvader,

I made a commit to fix the issue, I want to do a few more tests then close the issue and make a version with the fix.

Hi @jhofstee, There is one configuration issue when the two Ve.Can interfaces are enable that require different argument for the base topic for each mqtt-n2k instance to avoid conflict on mqtt.

Miliox, 2018-01-05:
Issue fix on version 0.1.4 of mqtt-n2k.

Miliox, 2018-01-19:
Hi @mpvader,
I talked I little bit with @fry. I think the best would be to change the name because I want the service to be separately addressable from other running services on venus and on the NMEA 2000.

But I also would need a unique number for the device name part from what @fry explained to me. There are unique numbers for MainInterface and vLink. There should one to be used by mqtt-n2k. What do you think?

Miliox, 2018-04-23:
@mpvader I not sure what exactly you mean by change the name, but now the default address of the service is 102 on version 0.1.5

Miliox, 2018-05-31:
Appear to be working fine since 0.1.5, so I will close this issue.

closed 2018-05-31 by Miliox