victronenergy/venus #239: gui ESS reason codes: small improvements

mpvader, 2017-12-22:
- [ ] box should be moved forward
- [ ] change the color, red indicates an error. there is no error, lots of systems will show these reason codes every day
- [ ] also on a more general note the layout can be a bit improved I think, @jepefe  ?

[image]

Reason codes:

* #1: SoC is low
* #2: BatteryLife is active
* #3: BMS disabled charging
* #4: BMS disabled discharge
* #5: Slow Charge in progress (part of BatteryLife, see manual)
* #6: User configured a charge limit of zero.
* #7: User configured a discharge limit of zero.

jepefe, 2018-01-30:
I've made some changes and pushed to improve_overview_ess_codes branch. 
[image]

mpvader, 2018-01-30:
@izak, pls check and when you agree schedule this for inclusion for v2.12

closed 2018-02-01 by jhofstee