victronenergy/venus #243: Venus GX: Tank & IO in Device list despite setting of switch

mpvader, 2018-01-02:
Reported by Warwick in the dev-venus list. Izak pls verify, and fix if necessary. And let me know when the bug was introduced.

Hello.

I have noticed that all "Tank" and "Temperature sensors" appear on the Device List despite the setting of the switches in I/O menu.
If I toggle the switches on / off again in I/O and reboot as suggested, all "Tanks" and "Temperature sensors" are still visible in the Device List showing as "Not Connected".

Wishing everyone a Happy New Year!

Warwick.

mpvader, 2018-01-17:
@izak pls check this before v2.12 release

izak, 2018-01-18:
I cannot reproduce it on 2.12~23, turning the sensors off leaves them off. I have some things I want to try before I declare it not-broken but there is at least no obvious regression.

izak, 2018-01-18:
Okay, reproduced and confirmed. Seems to be some kind of race condition. Sometimes it happens, sometimes it doesn't. You don't even have to reboot. Just restarting dbus-adc makes them show up.

Nothing related to this has been touched since June 30th last year, so it's definitely not a new bug. It's an old one.

izak, 2018-01-18:
Okay, I know why it happens. It creates the dbus services on startup, but then deletes them shortly afterwards. If the gui is already alive at this point, this event causes it to be added to the list, where it remains.

If you run dbus-spy while restarting dbus-adc, you see them flash on and off the screen quickly.

closed 2018-01-19 by jhofstee