victronenergy/venus #245: mqtt-rpc: dup exit code 3 (no response) on an empty system

mpvader, 2018-01-03:
That error should not be shown as an error to the user. Instead he should just see an empty list, or a sentence 'no devices found' orso.

Perhaps catch the exit code 3 in mqtt-rpc and change it into something else?

mauritsvdvijgh, 2018-01-03:
Steps to reproduce this/site to reproduce is?

mpvader, 2018-01-03:
Right now Martin Desk / https://vrm.victronenergy.com/installation/8072/dashboard has no devices connected.

mpvader, 2018-01-03:
actually, the issue might be a bit different: the empty list is shown:

[image]

But, shouldn't VRM instead have shown the DUP error? Now its hiding it.

And ofcourse DUP and/or mqtt-rpc then shouldn't give this false error.

Corresponding mqtt-rpc log:

```
2018-01-03 10:26:21.438972500 INFO:__main__:Ping received, sending pong back
2018-01-03 10:26:22.536384500 INFO:__main__:Vuplist received, starting vup to scan for devices
2018-01-03 10:26:22.549171500 INFO:streamcommand:Thread started for running ['vup', '--xml', '-vv', '--canbus', 'socketcan:can0']
2018-01-03 10:26:25.045539500 INFO:streamcommand:Thread finished for running ['vup', '--xml', '-vv', '--canbus', 'socketcan:can0']
2018-01-03 10:26:25.105781500 INFO:streamcommand:Command ['vup', '--xml', '-vv', '--canbus', 'socketcan:can0'] execution completed ok. Exitcode 0
2018-01-03 10:26:25.192268500 INFO:__main__:Duplist received, starting dup to scan for devices
2018-01-03 10:26:25.198585500 INFO:root:Calling serial starter to stop ttyUSB0
2018-01-03 10:26:25.396614500 INFO:root:Calling serial starter to stop ttyO2
2018-01-03 10:26:26.645301500 INFO:root:Calling serial starter to stop ttyO0
2018-01-03 10:26:26.867439500 INFO:streamcommand:Thread started for running ['dup', '--xml', '-v', '-s', '/dev/ttyUSB0 /dev/ttyO2 /dev/ttyO0']
2018-01-03 10:26:35.375923500 INFO:streamcommand:Thread finished for running ['dup', '--xml', '-v', '-s', '/dev/ttyUSB0 /dev/ttyO2 /dev/ttyO0']
2018-01-03 10:26:35.404670500 INFO:streamcommand:Command ['dup', '--xml', '-v', '-s', '/dev/ttyUSB0 /dev/ttyO2 /dev/ttyO0'] execution completed ok. Exitcode 3
2018-01-03 10:26:35.413795500 INFO:root:Calling serial starter to start ttyUSB0 again
2018-01-03 10:26:35.488716500 INFO:root:Calling serial starter to start ttyO2 again
2018-01-03 10:26:35.607093500 INFO:root:Calling serial starter to start ttyO0 again
```

mauritsvdvijgh, 2018-01-03:
When vup or duplist return an error it is caught and an empty list is returned for vup or duplist respectively.

What is then returned for device_list is vup and duplist combined (so [duplist + vuplist] [vuplist] when duplist has an error, [duplist] when vuplist has an error and [] when both have an error.

We are planning to change the list into an object with information about the status of vuplist/duplist and partial updates (send vup devices after vuplist has finished instead of waiting for both vup and duplist to finish before sending the complete list of devices). This works more or less the same for vebuslist by the way.

Alex suggested the following (for sending from vrm_mqtt_rpc to VRM):

Frame 0:
```
{
    progress: 0,
    success: null,
    partial: true
    devices: [ ]
}
```
Frame 1:
```
{
     progress: 50,
     success: null,
     partial: true
     devices: [ vup_devices ]
}
```
Frame 2:
```
{
    progress: 100,
    success: true,
    partial: false
    devices: [ vup_devices, dup_devices ]
}
```

mpvader, 2018-01-03:
That a vup error doesn't prevent updating a dup device and vice versa is good. But, also the error needs to be shown. No need to do that in a popup or modal.

Showing percentages, or 0, 50, 100 is something you shouldn't do. Its also not necessary. As long as there is a state that changes now and then while going through the different steps is good enough.

The different states can be:

1. Opening connection to device (this would be ping/pong)
2. Listing VE.Can devices (this would be vuplist)
3. Listing VE.Direct devices (this would be duplist)

Updating the list in partial updates can be done, but if it leads to more work, more code to write & more to maintain, I don't want it. Showing states is enough.

And perhaps there is a state that comes before above list, where the vrm backend contacts the celery worker? (not sure if we should show it as a state, or just show an error & number when its unavailable)

mpvader, 2018-01-17:
this has been fixed in various updates to both mqtt-rpc and the code on the vrm portal.

closed 2018-01-17 by mpvader