victronenergy/venus #246: mqtt-rpc: allow updating a device when in boot-mode?

mpvader, 2018-01-03:
Right now the code doesn't allow updating a device when in boot mode.

Which can be rather frustrating.

The reason behind this is that settings migration (obviously) doesn't succeed when updating a device from bootmode.

Only some long running models don't have built-in settings migration (vedirect mppts for example.. ). There must be a list somewhere.

Perhaps just add a warning and let the user continue regardless?

mpvader, 2018-01-10:
To continu with this, first @martinbosma will find out what `settingsMigration="True"` means in the vup & dup responses.

martinbosma, 2018-01-10:
For vup `settingsMigration` is set to `True` when the device is not in boot-mode && the productid is set && the productid != 0xFFFF.

For dup 'settingsMigration' is set to `True` when the device is not in boot-mode.

mpvader, 2018-07-12:
Considerations:

* It would be nice if the customer can self-manage himself out of a situation where the device is in bootloader.
* For devices that need settings migration (mppts, and others) a warning is necessary.
* For other devices, there should not be a warning, since it will only be confusing.
* For the MPPTs, that will no longer require a migration once TFS is implemented, we will need to always show a warning: once in bootloader modus, you don't know if there are settings in the device or not. 
* VictronConnect handles this situation by showing a warning after an update failure.

Conclusion:

We'll have a list of productids for which we need to show the warning, in the frontend. It will be a static list: new products will normally not require external settings migration.

@rvanzundert will provide a list of productids. See similar document [here on the internal wiki](https://wiki.victronenergy.com/wiki/rend/autovupanddup#products_that_need_migrating_settings).

mpvader, 2018-07-12:
Teun, we need the VRM frontend to show a warning in case the productid is in above mention list AND the device is in bootloader mode. Please propose the text/wire frame. Planning: if possible released before end of next week, but not critical.

Also we need to add `-i` to the dup call and `-k` to vup, hardcoded, in mqtt-rpc.

mauritsvdvijgh, 2018-07-12:
mqtt-rpc changes seem trivial, so that is not an issue.

In the vrm frontend we can show a red/orange "update" button which when clicked will display a confirmation dialog with a text like "Warning: You are about to update a device in boot loader mode, this will result in a total reset of the settings of this device" and close/confirm buttons.

From the vrm backend we will need to send the list of devices for which to show the warning if they are in boot loader mode.

No changes needed in vrm-mqtt-rpc.

mpvader, 2018-07-12:
> show a red/orange "update" button
yes, great idea! Orange rather than red please.

> a text like "Warning: You are about to update a device in boot loader mode, this will result in a total reset of the settings of this device"

"Warning: this device is now in bootloader mode. And in some cases this might mean that all settings are reverted back to factory default. Please verify the settings after completing the update."

mauritsvdvijgh, 2018-07-12:
The approved design for completeness' sake:

[image]
[image]

mpvader, 2018-07-13:
Some info to help you debugging:

To put a vup device in bootloader mode, use `vup -n 0x21 -u`. The -u does the trick, -n is the network address.

To put a dup device in bootloader mode, start an update, and press ctrl-c halfway: `dup -s /dev/ttyO0 -f ./BMV-702\ v3.08.dup`. That dup file is available on [Groningen Hub-1](https://vrm.victronenergy.com/installation/2931/settings/firmware-update).

mauritsvdvijgh, 2018-07-15:
What do you want to show the user in case the device is in boot mode, but it's not on the list of devices that use settings migration?

Some options:
1. orange button, warning sign, boot-mode, warning
2. orange button, warning sign, boot-mode
3. green button, warning sign, boot-mode
4. green button

I will implement the no. 3 version for now.

mauritsvdvijgh, 2018-07-15:
What format are the ids that vup/dup returns for: `0xA101, 0xA040..0xA07F`

decimal representation of the hexadecimals or hexadecimals without 0x or something else?

for `VE.Can Skylla-i chargers (0x0104..0x0107)` it returns strings `"104" "105" "106" etc.`

rvanzundert, 2018-07-16:
dup (-x) on 0xA040..0xA07F:
```
<?xml version="1.0" encoding="UTF-8"?>
<duprun version="1.0.14">
<discovered>
<device port="COM2" serial="HQ1747692YN" mode="app" description="SmartSolar Charger MPPT 100/20 48V" id="A060" version="1.34" updatable="True" settingsMigration="True"></device>
</discovered>
</duprun>
```

dup to stdout (without -x):
```
VE.Direct Updater

| port  | serial                       | mode | id   | description                      | version  |
| COM2  | HQ1747692YN                  | APP  | A060 | SmartSolar Charger MPPT 100/20 48V | 1.34     |

To update the first device in the above list, use the following command:

```

mpvader, 2018-07-16:
Both vup & dup return hexadecimals without 0x.

mauritsvdvijgh, 2018-07-16:
This change is live on production so this issue can be closed.

mpvader, 2018-07-18:
Tested, all works well. And docs have been updated:

https://github.com/victronenergy/venus-private/wiki/mqtt-rpc-and-vrm#updatable-information--handling-devices-in-bootloader

mauritsvdvijgh, 2018-07-19:
@mpvader /close

closed 2018-07-19 by mpvader