victronenergy/venus #248: mqtt: from some machines the first messages after subscription have payload with {value: null}

sztupi, 2018-01-04:
A weird issue we're having: 
On a new connection (with random-generated client session id, so there should be no previous state) we subscribe to a few channels of a portal, and we expect the first notifications to have value of the last known value of the topic.
This works exactly as expected on our development machines, but running the same code (same docker binary, same parameters), on production system (EC2 nodes on AWS) result in the first messages to be '{"value": null}';
Earlier (~19th of Dec, 2017), no subsequent notification arrived if the underlying value hasn't changed (e.g. it was consistently 0.0); today (4th of Jan, 2018) we got immediate notifications for all subtopics with the proper value.

Did anyone seen this or similar behaviour?

mpvader, 2018-01-18:
@wiebeytec pls look into this.

mpvader, 2018-01-18:
For chatting with sztupi and his colleagues, see venus-moixa in slack. We're running a project together.

wiebeytec, 2018-01-24:
@sztupi did you see my messages in Slack? If possible, I would like to do some interactive debugging.

wiebeytec, 2018-01-31:
We discussed this in Slack. There are non-ascii bytes included in topic path names.

@sztupi said:

> I'm about 90% sure it's the Haskell->libmosquitto FFI binding we use; actually I traced the problem to a few lines of code that probably are the culprit. will try to reproduce / fix / send a pull request to the binding in next few hours. will update you (and will close the issue it solves it) as I know more.

sztupi, 2018-02-02:
As mentioned above, the issue was with the C bindings (not marshalling strings properly), nothing to do where the code was running from.

closed 2018-02-02 by sztupi