victronenergy/venus #252: vup: update failures

mpvader, 2018-01-19:
This is now the 2nd or 3rd time, during development of mqtt-rpc, that I see a vup update failing on Groningen Hub-1, while updating the MPPT on address 0x24.

Retrying it has always solved it.

Which is a bit of a problem, because all settings are lost in the process.

```
# vup -k -n 0x24 -f ./Solar\ Charge\ Controller\ MPPT\ 150-70\ v2.04_108.vup --xml 

<vuprun version="2.0.5">
<message type="normal">Querying devices</message>
<message type="normal">Device found, migrating settings.</message>
<message type="normal">Retrieved settings, begin update.</message>
<message type="normal">UDF is ready to accept update.</message>
<message type="normal">Sending file ./Solar Charge Controller MPPT 150-70 v2.04_108.vup</message>
<progress level="1" />
<progress level="2" />
<progress level="3" />
<progress level="4" />
<progress level="5" />
<progress level="6" />
<progress level="7" />
<progress level="8" />
<progress level="9" />
<progress level="10" />
<progress level="11" />
<progress level="12" />
<progress level="13" />
<progress level="14" />
<progress level="15" />
<progress level="16" />
<progress level="17" />
<progress level="18" />
<progress level="19" />
<progress level="20" />
<progress level="21" />
<progress level="22" />
<progress level="23" />
<progress level="24" />
<progress level="25" />
<progress level="26" />
<progress level="27" />
<progress level="28" />
<progress level="29" />
<progress level="30" />
<progress level="31" />
<progress level="32" />
<progress level="33" />
<progress level="34" />
<progress level="35" />
<progress level="36" />
<progress level="37" />
<progress level="38" />
<progress level="39" />
<progress level="40" />
<progress level="41" />
<progress level="42" />
<progress level="43" />
<progress level="44" />
<progress level="45" />
<progress level="46" />
<progress level="47" />
<progress level="48" />
<progress level="49" />
<progress level="50" />
<progress level="51" />
<progress level="52" />
<progress level="53" />
<progress level="54" />
<progress level="55" />
<progress level="56" />
<progress level="57" />
<progress level="58" />
<progress level="59" />
<progress level="60" />
<progress level="61" />
<progress level="62" />
<progress level="63" />
<progress level="64" />
<progress level="65" />
<progress level="66" />
<progress level="67" />
<progress level="68" />
<progress level="69" />
<progress level="70" />
<progress level="71" />
<progress level="72" />
<progress level="73" />
<progress level="74" />
<progress level="75" />
<progress level="76" />
<progress level="77" />
<progress level="78" />
<progress level="79" />
<message type="error" module="UPD" timestamp="25419">RegId 0x0012 nacked with 8400</message>
<message type="error">ERROR (8400)</message>
<message type="error">ERROR (8400)</message>
<message type="normal">Update end send</message>
</vuprun>
```

jhofstee, 2018-01-22:
Error 8400 means the bootloader was expecting data, but didn't receive it within 15seconds

mpvader, 2018-01-23:
won't fix for now.

closed 2018-01-23 by mpvader