victronenergy/venus #254: velib_utils

mpvader, 2018-01-29:
Let `get_vrm_portal_id()` use the contents of `/data/venus/uniqueid` rather than getting the mac address itself.

That uniqueid file is created on all machines, see here for how its done: https://github.com/victronenergy/meta-victronenergy/tree/master/meta-bsp/recipes-bsp/machine-runtime-conf/files

mpvader, 2018-01-29:
Note that relying on only that file, without falling back to a syscall, makes vrmlogger depend on a working data partition.

which will be a problem for raspbian systems, and can also be a problem for broken data partitions,
 though how those systems behave is still undetermined in general.

all would easily be solved by fallback to the syscall as @izak pointed out.

mpvader, 2018-03-20:
looks good. pls update all projects (and check how big of a step forward for the lib it is for those projects to determine required testing).

There are more than you'd think 😄 

https://github.com/search?q=org%3Avictronenergy+get_vrm_portal_id&type=Code

mpvader, 2018-07-12:
@izak pls make a list of project, and checkbox which ones now use /data/venus/unique-id, and which don't, to close this issue.

izak, 2018-10-30:
Products that now use /data/venus/unique-id are:

* mqtt-rpc
* dbus-systemcalc-py
* vrmlogger
* dbus-mqtt

These were all handled in v2.20~20. Other products that use `velib_python` do not use `get_vrm_portal_id`.

closed 2018-10-30 by izak