victronenergy/venus #258: Localsettings Startup Performance

3dGrabber, 2018-02-01:
I made performance benchmarks of `localsettings` during startup on the CCGX.
I found that `/usr/bin/python` takes a lot of time to start up the first time (~15 secs), whereas the performance of the script, itself (~2s) seems satisfactory. When the `localsettings` service is restarted at a later time, python startup is much faster.

### at boot time:

    starting python:       ~15 sec !
    init localsettings.py:  ~2 sec

### localsettings service restarted:

    starting python:        ~2 sec
    init localsettings.py:  ~2 sec

My guess is, that `/usr/bin/python` needs that time to load its libs from "disk", and when run a second time the libs are already cached in RAM.
This, or the slowdown is because the system is under heavy load at startup (probably IO limited).
The latter seems less probable, since the script itself also reads from disk and doesn't seem to be much slower at boot time.

mpvader, 2018-02-14:
@3dGrabber pls add the boottime graph that was made into this issue as well as a short summary of anything else important that we discussed (above timing was measured without pyc files already being present, or something like that?)

and then close the issue, as the conclusion was that that we can better improve boot time in a different way

3dGrabber, 2018-02-14:
For future reference: 

```
I made performance benchmarks of localsettings during startup.
I injected probes at the following locations:

- in the services' run script, just before the python interpreter is called.  ("starting service")
- at the first line localsettings.py, in front of all imports. ("starting script")
- in localsettings.py where the main function is called and the script starts to do actual work. ("running script")
- after initialization of localsettings.py (read/parse setting files), before it starts asynchronous (reactive) operation ("initialized")

## TEST 1 ######################

root@ccgx:/service# cat /log/localsettings_benchmark

starting service: Thu Feb 1 09:14:33 UTC 2018
starting script: 2018-02-01 09:14:48.695768
running script: 2018-02-01 09:14:48.885282
initialized: 2018-02-01 09:14:50.744169

root@ccgx:/service# svc -d localsettings
root@ccgx:/service# svc -u localsettings
root@ccgx:/service# cat /log/localsettings_benchmark

starting service: Thu Feb 1 09:16:28 UTC 2018
starting script: 2018-02-01 09:16:29.712583
running script: 2018-02-01 09:16:29.733793
initialized: 2018-02-01 09:16:30.038847

## TEST 2 ######################

root@ccgx:/service# cat /log/localsettings_benchmark

starting service: Thu Feb 1 09:27:48 UTC 2018
starting script: 2018-02-01 09:28:03.846193
running script: 2018-02-01 09:28:04.052309
initialized: 2018-02-01 09:28:06.244539

root@ccgx:/service# svc -d localsettings
root@ccgx:/service# svc -u localsettings
root@ccgx:/service# cat /log/localsettings_benchmark

starting service: Thu Feb 1 09:29:56 UTC 2018
starting script: 2018-02-01 09:29:58.028810
running script: 2018-02-01 09:29:58.045381
initialized: 2018-02-01 09:29:58.336519

## TEST 3 ######################

root@ccgx:/service# cat /log/localsettings_benchmark

starting service: Thu Feb 1 09:35:19 UTC 2018
starting script: 2018-02-01 09:35:34.358440
running script: 2018-02-01 09:35:34.578300
initialized: 2018-02-01 09:35:36.472403

root@ccgx:/service# svc -d localsettings
root@ccgx:/service# svc -u localsettings
root@ccgx:/service# cat /log/localsettings_benchmark

starting service: Thu Feb 1 09:36:38 UTC 2018
starting script: 2018-02-01 09:36:39.892584
running script: 2018-02-01 09:36:39.912268
initialized: 2018-02-01 09:36:40.293768
```

3dGrabber, 2018-02-14:
Bootchart made by @mansr on a beaglebone:   

"you can tell where localsettings is ready from start-gui.sh ending and gui starting"

[bootchart.pdf](https://github.com/victronenergy/venus/files/1723948/bootchart.pdf)

3dGrabber, 2018-02-14:
the conclusion was that that we can better improve boot time in a different way

closed 2018-02-14 by 3dGrabber