victronenergy/venus #261: check carlo gavazzi EM300 & ET300 compatibility

mpvader, 2018-02-05:
Carlo Gavazzi has a new meter, the EM330. The difference between this, and the one we already have, the ET340, is that the EM330 is CT based, and the ET340 is shunt based. `Direct Connect` in CG-speak.

The difference between the EM330 and the EM24 which we already support is that the EM330 is 3 units wide instead 4. Also the new one is more cost effective.

Both share the same modbustcp template. The identifier will be different though. Needs checking!

Also, the UK CG sales rep warned me that the EMs don't have a negative kWh counter per phase. Just like the EM24 doesn't, which makes it insuitable to measure grid on L1 and PV on L2.

[image]

Docs here:
[EM300 and ET300-V.2.rev 11.pdf](https://github.com/victronenergy/venus/files/1695884/EM300.and.ET300-V.2.rev.11.pdf)

mpvader, 2018-02-05:
I've asked the customer that proposed this meter to get one and wire it up, so we can login remotely and check that it works well.

mpvader, 2018-02-05:
Note that the ET330 does not exist (yet), as confirmed by the UK CG sales rep.

mpvader, 2018-03-15:
Test system with meter connected:
https://vrm.victronenergy.com/installation/18959/diagnostics

mpvader, 2018-03-15:
Meter type is EM330-DIN.AV5.3.H.S1.X.

What needs to be done is checking what ID that meter has, and then (most probably) add that the dbus-cgwacs to make it all work.

izak, 2018-03-16:
We should already detect the 330 as a member of the [family][1], the code already detects that entire spectrum between 330 and 345 as the same kind of meter.

The trouble here seems to be that the meter isn't replying, we get a timeout. Haven't digged in more detail than that, this is how far I get: It seems to timeout before we even get a DeviceId response.

[1]: https://www.galoz.co.il/wp-content/uploads/2014/11/EM341-Modbus.pdf

mpvader, 2018-07-12:
No response from the customer that wanted this; closing for now.

closed 2018-07-12 by mpvader

warwickchapman, 2022-07-28:
Hi Both

Our Carlo supplier here in SA say they’re unable to get the EM24 CT version and it is now discontinued and we should use the EM330 which they described as an EM24 inside a smaller housing. From what is described above, this appears not to be he case. 

Nonetheless, I am interested in whether this meter is back on the radar given the shortages of meters from Carlo and uncertainty about when e new EM530 CT version will be available. 

Warwick.

izak, 2022-07-29:
Hi @warwickchapman ,

You're replying on a really old issue here.

The EM300 series (including the ET300 series) are, to my knowledge, all the same in terms of how they work. The reason I am not particularly fond of the EM300 series is because the energy counting is different (or "wrong" in other words) from the EM24. But they should all work, and if they don't, then we can make a plan so that they do.

warwickchapman, 2022-07-29:
OK, apologies for resurrecting the old issue.  Your answer is noted.  If we
have an issue, we will advise.

Warwick Bruce Chapman | +27 83 7797 094 | http://wa.rwick.com