victronenergy/venus #262: USB485-CPT isolated cable

mpvader, 2018-02-13:
[Datasheet of the device](https://github.com/victronenergy/venus/files/1719743/Product.sheet.-.USB485-CPT.pdf)

### On a CCGX:
```
[    3.096923] usb 1-1.1: New USB device found, idVendor=1c40, idProduct=0477
[    3.097015] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.097045] usb 1-1.1: Product: USB485 Iso stick
[    3.097076] usb 1-1.1: Manufacturer: Hjelmslund Electronics
[    3.097106] usb 1-1.1: SerialNumber: STISO485
```

lsusb:
```
root@ccgx:~# lsusb
Bus 001 Device 003: ID 1c40:0477 EZPrototypes 
Bus 001 Device 002: ID 0424:2412 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

And then no tty is created.

I'll add results of plugging it in elsewhere (Venus GX and/or desktop pc), as soon as I have them

mansr, 2018-02-13:
What is the output of `lsusb -v -d 1c40:0477`?

mpvader, 2018-02-13:
```
# lsusb -v -d 1c40:0477

Bus 001 Device 003: ID 1c40:0477 EZPrototypes 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1c40 EZPrototypes
  idProduct          0x0477 
  bcdDevice            6.00
  iManufacturer           1 Hjelmslund Electronics
  iProduct                2 USB485 Iso stick
  iSerial                 3 STISO485
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 USB485 Iso stick
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0000
  (Bus Powered)
```

mpvader, 2018-02-13:
fyi @mansr http://www.hjelmslund.dk/Files/Note%20for%20Linux%20users.pdf

mansr, 2018-02-13:
`echo 1c40 0477 >/sys/bus/usb-serial/drivers/ftdi_sio/new_id` will make the driver bind to this device. For a permanent solution, the ftdi_sio driver needs to be patched. Then you'll need to tell serial-starter about this device. That can be done with a udev rule:
```
ACTION=="add", ENV{ID_BUS}=="usb", ENV{ID_VENDOR_ID}=="1c40", ENV{ID_MODEL_ID}=="0477", ENV{VE_SERVICE}="cgwacs"
```

mpvader, 2018-02-14:
OK I've confirmed that the cable works (had to modprobe ftdi_sio first and then insert those ids into the driver). 

I've arranged for one of those cables to be sent to you, @mansr and as discussed pls include this driver change in the coming kernel updates.

as well as look into defining a group of programs that should be started when the link is determined to be rs485 (cgwacs, redflow, and there are some drivers being worked on right now)

mansr, 2018-02-26:
Kernel patch is done. Serial-starter changes are pending inclusion in master.

Plugging this device into a running beaglebone makes all other USB devices disconnect briefly, probably due to a power surge. Everything recovers quickly, so it should not cause any problems.

mansr, 2018-03-02:
Kernel and serial-starter changes are done.

Plugging this device into a running beaglebone makes all other USB devices disconnect briefly, probably due to a power surge. Everything recovers quickly, so it should not cause any problems. No issues on ccgx.

mansr, 2018-03-02:
All required changes are in v2.20~10.

The power glitch only affects the beaglebone (Venus GX). The ccgx is fine.

closed 2018-03-02 by mpvader