victronenergy/venus #263: GPS receivers that support other GNSS systems are not supported

gsosna, 2018-02-13:
**PROBLEM:**
Venus GX currently only supports American GPS NMEA sentences. A device that is able to receive and combine multiple satellite systems is called GNSS. GNSS receivers support multiple systems such as:

- GPS (USA)
- GLONASS (Russian)
- Galileo (European)
- Beidou (Chinese).

Plugging in a GNSS receiver to Venus, shows up as "GPS Connected, but no GPS fix"

[image]

**REASON:**

GNSS devices are likely to send a slightly different set of NMEA sentences to indicate which system the location information is derived from. While "pure" GPS (USA) will issue a GPGGA sentence with its location, a GNSS receiver will use the GNGGA command to indicate that the location is derived from multiple satellite systems. Venus appears to not understand the GNGGA sentence. After I changed configuration of my GNSS receiver to only send GPGGA sentences, Venus reports the location correctly.

**TECHNICAL:**

On Venus GX shell I ran: 

/opt/victronenergy/gps-dbus/gps_dbus -vvvvvvvvvvv --banner --dbus system --timeout 2 -s /dev/ttyACM0 -b 4800

to monitor what's going on and confirmed my suspicion:

gps_dbus 1.02
001.194 INF.VALUES: connected to dbus
001.195 INF.GPS: GNRMC,114613.00,V,,,,,,,,,,N
002.077 INF.GPS: GNGGA,114614.00,9304.52905,S,17122.22240,E,1,04,1.81,17.1,M,28.3,M,,
002.077 INF.GPS: GNGSA,A,3,03,22,26,32,,,,,,,,,6.37,1.81,6.10
003.034 INF.GPS: GNRMC,114615.00,A,3246.52946,S,17122.22316,E,0.071,,130218,,,A
003.034 INF.GPS: GNVTG,,T,,M,0.071,N,0.131,K,A
004.033 INF.GPS: GNRMC,114616.00,A,3246.52970,S,17122.22361,E,0.317,,130218,,,A
004.033 INF.GPS: GNVTG,,T,,M,0.317,N,0.587,K,A
005.039 INF.GPS: GNRMC,114617.00,A,3246.53016,S,17122.22426,E,0.037,,130218,,,A
005.041 INF.GPS: GNGGA,114617.00,3246.53016,S,17122.22426,E,1,11,0.98,22.6,M,28.3,M,,
005.041 INF.GPS: GNGSA,A,3,03,22,26,32,31,14,01,25,,,,,1.96,0.98,1.70
005.041 INF.GPS: GNGSA,A,3,68,69,78,,,,,,,,,,1.96,0.98,1.70
005.041 INF.GPS: GPGSV,4,1,16,01,25,277,24,03,37,227,36,04,,,27,10,12,033,
005.041 INF.GPS: GPGSV,4,2,16,11,04,286,,14,54,120,31,16,16,004,22,22,56,244,38
005.041 INF.GPS: GPGSV,4,3,16,23,09,251,,25,23,133,25,26,39,023,35,31,74,155,28
005.041 INF.GPS: GPGSV,4,4,16,32,36,097,35,46,18,068,,48,21,065,,51,01,083,
005.041 INF.GPS: GLGSV,3,1,09,67,11,052,20,68,61,080,25,69,53,202,31,70,01,222,
006.030 INF.GPS: GNRMC,114618.00,A,3246.53038,S,17122.22453,E,0.021,,130218,,,A
006.030 INF.GPS: GNVTG,,T,,M,0.021,N,0.039,K,A
006.031 INF.GPS: GPGSV,4,3,16,23,09,251,,25,23,133,25,26,39,023,35,31,74,155,28
006.032 INF.GPS: GPGSV,4,4,16,32,36,097,35,46,18,068,,48,21,065,,51,01,083,
006.032 INF.GPS: GLGSV,3,1,09,67,11,052,19,68,61,080,25,69,53,202,31,70,01,222,
007.041 INF.GPS: GNRMC,114619.00,A,3246.53056,S,17122.22474,E,0.075,,130218,,,A
007.041 INF.GPS: GNVTG,,T,,M,0.075,N,0.139,K,A
007.042 INF.GPS: GNGGA,114619.00,3246.53056,S,17122.22474,E,1,12,0.91,23.1,M,28.3,M,,
007.042 INF.GPS: GNGSA,A,3,03,22,26,32,31,14,01,25,,,,,1.68,0.91,1.41
007.042 INF.GPS: GNGSA,A,3,68,69,78,67,,,,,,,,,1.68,0.91,1.41
007.042 INF.GPS: GPGSV,4,1,16,01,25,277,24,03,37,227,36,04,,,28,10,12,033,17
007.042 INF.GPS: GPGSV,4,2,16,11,04,286,,14,54,120,31,16,16,004,22,22,56,244,38
007.042 INF.GPS: GPGSV,4,3,16,23,09,251,,25,23,133,26,26,39,023,35,31,74,155,28
008.032 INF.GPS: GNRMC,114620.00,A,3246.53058,S,17122.22481,E,0.014,,130218,,,A
008.034 INF.GPS: GNGLL,3246.53058,S,17122.22481,E,114620.00,A,A
009.041 INF.GPS: GNRMC,114621.00,A,3246.53059,S,17122.22487,E,0.036,,130218,,,A
009.041 INF.GPS: GNVTG,,T,,M,0.036,N,0.066,K,A
010.030 INF.GPS: GNRMC,114622.00,A,3246.53060,S,17122.22496,E,0.073,,130218,,,A
010.030 INF.GPS: GNVTG,,T,,M,0.073,N,0.135,K,A
010.032 INF.GPS: GLGSV,3,2,09,77,01,149,,78,48,147,28,79,25,332,,80,19,329,
010.032 INF.GPS: GLGSV,3,3,09,85,01,230,
010.032 INF.GPS: GNGLL,3246.53060,S,17122.22496,E,114622.00,A,A
011.042 INF.GPS: GNRMC,114623.00,A,3246.53058,S,17122.22494,E,0.093,,130218,,,A
011.042 INF.GPS: GNVTG,,T,,M,0.093,N,0.173,K,A
012.031 INF.GPS: GNRMC,114624.00,A,3246.53056,S,17122.22490,E,0.070,,130218,,,A
012.031 INF.GPS: GNVTG,,T,,M,0.070,N,0.130,K,A
012.032 INF.GPS: GLGSV,3,1,09,67,11,052,21,68,61,080,27,69,53,202,31,70,01,222,

After I changed configuration of my GNSS receiver to only send GPGGA sentences, Venus reports the location correctly.

mpvader, 2018-05-02:
hey @gsosna sorry for the late reply. Are you interested in making a fix in our code for it?

mpvader, 2018-05-02:
ps; code is here: https://github.com/victronenergy/dbus_gps

and I would then give you access to the lib thats not public (velib)

malcolmh, 2018-05-06:
This can be fixed if instead of matching to "GPGGA" & "GPRMC", match to "..GGA" & "..RMC".

malcolmh, 2018-05-25:
gps.c has now been modified to work with any talker ID. Now any xxGGA & xxRMC will be decoded. Also I added a parser for the new xxGNS sentence.

mpvader, 2018-05-25:
Many thanks again @malcolmh!

Your commits have been included in dbus_gps release v1.03. Which is currently scheduled to be added into the next build of Venus.

closed 2018-05-30 by jhofstee

mpvader, 2018-05-30:
v2.20~33 has the new dbus_gps version and is now being built.