victronenergy/venus #267: ccgx reboots using Remote Console

mpvader, 2018-02-24:
Note that it seems fixed by using the 4.9 kernel (tested with v2.20~5).

The facts so far:
- using the gui on the device itself works fine; no reboots
- using the gui via Remote Console, both on VRM and LAN, triggers a reboot. 
- nothing shows up in the logs.
- just opening remote console, just enjoying the overview, not pressing any buttons, doesn't trigger the reboot.
- neither does a single key press (which wakes up the backlight)
- then pressing one more is enough to make it reboot. Sometimes immediately, sometimes it takes a few seconds and/or a few more key presses.
- disconnecting all equipment doesn't make a difference
- verified to happen on two devices, `A du Rand Home` and #650 in level3support
- I have heard (some) more complaints that might be the same issue (from Anco)

Affected versions:
- I saw it happening on v2.12 and v2.10, multiple times (A du Rand system)
- #650 customer in level3support [reported it happening on v1.74](https://www.victronenergy.com/live/ccgx:ccgx_faq#comment-3750207881)

Next step is that I'm trying to get one connected to the serial console and then make it happen again. Just to make sure. In the mean time hope & verify that 4.9 really fixes it.

mpvader, 2018-02-24:
to add: on those two systems it happens every single time you open Remote Console and press some buttons. Its impossible to use the Remote Console for anything other than looking at whatever page it is showing when you login.

jhofstee, 2018-07-06:
on the 'A du Rand home' this still happens every time. Also with v2.20~40 it happens directly when opening remote console. It seems as if power supply resets, but that doesn't seem to make any sense.

jhofstee, 2018-07-10:
using a different wifi USB dongle solves this is on 'A du Rand home'

jhofstee, 2019-03-20:
This remains a weird issue, but since it can be solved by replacing the hw, closing this for now. It seems to be some odd case.

closed 2019-03-20 by jhofstee