victronenergy/venus #276: mqtt-rpc: status updates not working on nanopi groningen

vejpasop, 2018-04-10:
On Groningen EasySolar Nanopi, running v2.20~16 with extra nanopi changes. mqtt-rpc v1.05.

Reproducible on this system.

Hitting F5 on the browser allows to retry. Does that mean that the communication to the users browser has a problem, whilst communication between mqtt backend and venus devices works ok, since it releases the lock?

Also we tried a firmware update scan, similar issue: stuck at `Starting querying devices... `.

The download veconfigure settings process did finish OK on the Venus device itself, see log. But, the file was not downloaded by the browser.
```
2018-04-10 07:03:24.721886500 INFO:__main__:Ping received, sending pong back
2018-04-10 07:03:25.709078500 INFO:__main__:Vebuslist received, starting VebusList.
2018-04-10 07:03:27.983390500 INFO:__main__:Ping received, sending pong back
2018-04-10 07:03:28.923530500 INFO:__main__:Starting mk2VscThread
2018-04-10 07:03:28.940777500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor starting... =====
2018-04-10 07:03:28.945898500 INFO:dbusmonitor:===== Search on dbus for services that we will monitor finished =====
2018-04-10 07:03:28.968968500 INFO:vrmmk2vsc:read_settings
2018-04-10 07:03:29.055071500 INFO:dbusmonitor:Found: com.victronenergy.mk2vsc, scanning and storing items
2018-04-10 07:03:29.121049500 INFO:dbusmonitor:       com.victronenergy.mk2vsc has device instance 0
2018-04-10 07:05:43.240300500 INFO:__main__:Exiting mk2VscThread
2018-04-10 07:05:43.244333500 INFO:dbusmonitor:com.victronenergy.mk2vsc disappeared from the dbus. Removing it from our lists
2018-04-10 07:05:43.304632500 INFO:__main__:Filedownload announce received. Starting file sender
2018-04-10 07:05:43.307708500 INFO:file_mgmt:FileSender initialized. Sending '/data/vrmfilescache/028191859555_inst_ttyS3_Interfaces_Mk2_Tunnel.rvsc' in 1 chunks
2018-04-10 07:05:43.407286500 INFO:file_mgmt:Chunk 0 acked by remote. That makes for the complete file.
2018-04-10 07:05:43.410140500 INFO:file_mgmt:FileSender object done, marking as expired
```

Meanwhile, the info on VRM is stuck on `Querying service`. That string btw is not one of the known ones, what kind of translation is going on? Perhaps better to just use the ones as sent by the Venus device; and if they need to be better, fix them there instead of translating them.
[image]

While trying to reproce it on a CCGX, running v2.20~19, also mqtt-rpc v1.05, we saw some other strange things, and were not able to reproduce the same behaviour as above:

Showing finished right after pressing the download button (after a previous download):
[image]

And also on time 'Reading settings data' showed up, but far to later.

And the first time it was showing 'waiting for status updates' for a too long time.

But, we didn't manage to reproduce the 'Querying service` being shown too long.

And even though the status updates didn't work, the settings file was downloaded by the browser in the end.

And scanning firmware update worked fine, including status updates in the browser.

mauritsvdvijgh, 2018-05-02:
`Hitting F5 on the browser allows to retry. Does that mean that the communication to the users browser has a problem, whilst communication between mqtt backend and venus devices works ok, since it releases the lock?`

At first I could not reproduce it because of something else, I just reproduced it and it seems that you are correct here.

`Meanwhile, the info on VRM is stuck on Querying service. That string btw is not one of the known ones, what kind of translation is going on? Perhaps better to just use the ones as sent by the Venus device; and if they need to be better, fix them there instead of translating them.`

It comes out of the frontend, they are also translated in the frontend. Could switch to the messages from the venus device, but we had that before and when asked to fix it we implemented this, disadvantage of using the text straight from the venus device is no translation (in vrm frontend). The querying device/service case specifically is before a message is received from the device.

Alex will take a look.

mpvader, 2018-07-11:
@mauritsvdvijgh: status?

thlassche, 2018-07-11:
@mpvader I just checked, the portal id of that site is: `028191859555_old_1529492830_5b2a355eaeaba` Which looks like the one of a swapped CCGX, before diving deeper in to this, is it possible that this type of portal id's is blocked in the MQTT auth layer (@wiebeytec)? That would explain the stuck in querying devices state (since we cannot detect that MQTT messages are blocked in the auth bridge)

thlassche, 2018-07-11:
@mpvader Nvm, I found the root cause, testing a fix now

thlassche, 2018-07-11:
@mpvader This one now correctly shows that the site communication is down, which is correct because it is not online I think

mpvader, 2018-07-11:
Ok! I’ll close the issue.

The bit about translating will resurface again by itself, in case important.

closed 2018-07-11 by mpvader