victronenergy/venus #282: mqtt-rpc: handle exceptions

mpvader, 2018-05-31:
Todo

- [ ] Decide how to show the Lynx Ion BMS to the user. Best would be that manual fw update is possible. Proposal: we make a blacklist of productids, and the website then shows 'remote firmware update not possible'. Also manual updates are then not possible. And for the batteries we make a new set of productids, where it works as its meant to be by Victron: a vup file per productid.
- [ ] Look into the Ion Control: add to the blacklist?

Details:

- Lynx Ion BMS: multiple models. one binary. But they do have different productids. And the issue is that the bootloader only accepts a binary that has a generic productid inside. the code that checks if the productid of the file being sent is valid, is *not* masking during its check. So, sending a file manually is possible (send the file with the generic productid). But using the firmware library is not possible: the generic file is listed under that generic productid, and therefor not matched.
- Lynx Ion + Shunts work OK btw: there are two models, but they share one productid, and also both models share one firmware binary.

mauritsvdvijgh, 2018-07-11:
We can implement a blacklist, do you want to be able to manage it and if so: how?

An option is to place the blacklist in the config for the worker, to add an id you would edit the config and restart the worker. There are other options if this is not sufficient.

For the Lynx Ion BMS I propose mapping the specific id to the generic id before doing the firmware library lookup. This would allow us to still offer the update through the firmware library for these products. No need for a blacklist it seems (in this case).

mpvader, 2018-07-11:
> We can implement a blacklist, do you want to be able to manage it and if so: how?

No need to manage it. Hardcoding is fine.

> For the Lynx Ion BMS I propose mapping the specific id to the generic id before doing the firmware library lookup. This would allow us to still offer the update through the firmware library for these products. No need for a blacklist it seems (in this case).

Yes, thats a nice solution. @martinbosma do you agree?

And, @martinbosma pls comment the serialnumbers (& mapping) here.

Regarding the Lynx Ion and Lynx Ion + Shunt: should we blacklist them as well? What happens if you update them remotely: do they switch the system off? In that case, lets black list those as well.

And what happens with the Lynx Ion BMS? Main contactor? And allow-to-charge and allow-to-discharge outputs?

martinbosma, 2018-07-12:
> Yes, thats a nice solution. @martinbosma do you agree?

Yes, that's fine

> And, @martinbosma pls comment the serialnumbers (& mapping) here.

Product id `0xA390-0xA39F` have been reserved for the Lynx Ion BMS, where `0xA390` is the generic id. Right now only `0xA391-0xA394` have been assigned to existing products. So the question is if it is fine to map the complete range to the generic id or only the ones that are used. 
- `0xA390`: Lynx Ion BMS Generic
- `0xA391`: Lynx Ion BMS 150A
- `0xA392`: Lynx Ion BMS 400A
- `0xA393`: Lynx Ion BMS 600A
- `0xA394`: Lynx Ion BMS 1000A

mpvader, 2018-07-14:
@mauritsvdvijgh pls block firmware updating for these product ids, and make for some solution in the UI:

* 0x0142 - Lynx Ion
* 0xA130 - Lynx Ion + Shunt
* 0x0140 - Ion Control
* 0xA391..0xA394 - Lynx Ion BMS 150, 400, 600 & 1000A.

Any more advanced things like remapping can be done later; before doing that we first need more input from the developers, and I don’t want to wait.

mauritsvdvijgh, 2018-07-16:
The blacklist is implemented and in use at the moment:

`        "firmware_update_blacklist": ["142", "140", "A130", "A391", "A392", "A393", "A394"]`

https://github.com/victronenergy/vrm-mqtt-rpc/commit/5f490d93a3f168482446b7e6fc12e63d4a98685e

mauritsvdvijgh, 2018-07-16:
This change is live on production so this issue can be closed.

mpvader, 2018-07-17:
Fixed and working well. Two new issues coming out of this:

1. Check if we can improve Remote updating of Lynx BMSes (Right now they are all blocked); input needed from developers before working on a better version. #298.

2. Add codes to the `Not updatable` text, so in the manual we can explain why. #296.

[image]

closed 2018-07-17 by mpvader