victronenergy/venus #284: Raspberry Pi3 onboard wifi not detected

rswhiting, 2018-06-21:
The documentation explicitly says that Pi3 onboard wifi is supported, but ifconfig only returns the eth0 and local loopback. I have the image for Linux raspberrypi2 4.9.50 installed. What am I missing? I can purchase an external wifi, but I got the Pi3 because it has onboard wifi.

jhofstee, 2018-07-05:
Do you have a Raspberry Pi 3 Model B+? Venus supports the Model 3 B, but there are some issues with the plus variant at the moment, see https://groups.google.com/forum/#!topic/victron-dev-venus/nZuRN96PJHQ

mpvader, 2018-07-12:
closing this for now. If it is still open, please specify exact type of raspberrypi 3 (a plus?) or use the mailinglist.

closed 2018-07-12 by mpvader