victronenergy/venus #286: Upgrading build server to 18.04 / 16.04

quinox, 2018-06-22:
I'd like to discuss the upgrade path for the CI server (`PowerBuilder`) with you fine people.

`PowerBuilder` currently is an Ubuntu 14.04 LTS machine. Back when the server was set up (in February 2017) this was the best choice as the build system wasn't fully compatible with 16.04 yet.

The upstream support for 14.04 LTS is ending at [the beginning of 2019](https://wiki.ubuntu.com/Releases), I'd like to upgrade the machine somewhere within the new few months.

My questions:
* Is the build process compatible with either 18.04 LTS or 16.04 LTS?
  * If yes: excellent!
  * If no (or unknown) there are three obvious paths to take:
    * I can wait until you confirm it's compatible with 18.04/16.04 and upgrade the machine to that release
    * I can upgrade the machine and you / me can fix any issues that pop up afterwards
    * I can test the build in a VM of 18.04/16.04 and fix any issues beforehand

jhofstee, 2018-06-22:
Hello quinox, it builds on my pc, which is running:

jeroen@blue:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.4 LTS
Release:	16.04
Codename:	xenial

Don't know about 18.04 LTS though..

mpvader, 2018-07-11:
18.04 is fine.

mpvader, 2018-07-11:
I mean: you have to try 18.04, and then find out.

mpvader, 2018-07-12:
no need for this issue; closing

closed 2018-07-12 by mpvader