victronenergy/venus #287: Network watchdog

mansr, 2018-06-22:
If internet connectivity is lost, a local malfunction may be the cause. It is thus desirable to detect such a condition and take appropriate action which may include rebooting the system. When connectivity is provided by a GSM/3G modem, this shall also be restarted.

### Detecting network loss
Using VRM TimeLastContact value avoids creating additional network traffic for the connectivity check. The logging interval setting must be taken into account. The check can be performed periodically by the watchdog daemon.

### Recovering
The Victron USB modem can be rebooted by stopping updates of its watchdog and waiting for it to reset. An onboard modem can be simply power cycled. A subsequent full system reboot ensures that a hung pppd or similar does not block communication.

### Timeout
To allow for an unreliable external network, a relatively long timeout shall be used. 60 minutes is likely a suitable choice. This timeout is in addition to the VRM logging interval, so the final amount can be significantly longer depending on VRM settings.

### To do
- [ ] Write helper script to reset modem
- [ ] Write watchdog script to check VRM status
- [ ] Compete to do list

mpvader, 2018-06-24:
Some additional points:

* it must be possible to disable the network watchdog. (I'm not sure if by default it should be enabled, or disabled).
* there is this: https://github.com/victronenergy/venus/issues/222. Could be done, implementation probably not so difficult; but yet more options can also be confusing for the user. I'm leaning towards not doing it. People that want reliable comms for Venus only can get it by getting the Venus GX. People that want reliable comms for more than just the Venus can get themselves a professional cellular router that includes a reboot function aka network watchdog.

mansr, 2018-07-20:
https://github.com/victronenergy/meta-victronenergy/tree/mans/watchdog

mpvader, 2018-07-21:
Thanks.

I do miss an on/off setting. Perhaps better, move the time-out to dbus and when 0 => disabled.

mpvader, 2018-10-31:
On off setting has been implemented. Waiting for inclusion in Venus.

mpvader, 2019-01-07:
The functionality has been added to Venus v2.30~8, with [this commit](https://github.com/victronenergy/meta-victronenergy/commit/4aebef034063b2c0d91f915dd18c97912d02b2b0). Now needs adding to the gui; and testing.

jhofstee, 2019-02-19:
[image]

comments?

mpvader, 2019-11-28:
Its in the ccgx manual

closed 2019-11-28 by mpvader