victronenergy/venus #288: When using an ET112 meter for a PV-inverter it can only be used on L1

izak, 2018-06-27:
For the ET112 meter we only publish values on /Ac/L1 because it is a single-phase meter. There is no way to adjust this. This means that when an ET112 is used to measure production of a PV-inverter, it can only be used for PV-inverters on L1. Ideally it should be possible to set a different phase for PV-inverters only, as it likely makes no sense for other applications.

mpvader, 2018-07-11:
I'm closing this, there are plenty other things todo. It will come back automatically if important.

mpvader, 2018-07-11:
ps; customers can buy a three phase meter if they really want this.

closed 2018-07-11 by mpvader

vincentwolsink, 2026-01-14:
I am actually running into this. My PV is single phase, but located on L3 because of self-consumption and net voltages being lower on L3. As a result VRM dashboard adds my solar production to the load on L1. I didn't notice any other side effects yet, besides the display error.

I am looking into https://github.com/victronenergy/dbus-cgwacs to open a PR, but my C++ isn't great. If I manage to produce something, would you be open to merging it? Or would I be wasting my time? :)