victronenergy/venus #289: Grid recharge issue at 0% SoC when MinSoC=5%

mpvader, 2018-07-05:
Reported by Simon:

When MinSoC is set to 5%, our observation is that if the battery drops to 0% SoC (vs 1%), the 'recharge' function does not operate. If external charging (e.g. DC MPPT) brings the battery back to 1% or higher, then the ESS 'Recharge' function starts to work again as expected to bring the charge above 5%. Have raised this with Izak and asked him if he can take a look at it.

mpvader, 2018-07-05:
(Is perhaps already fixed in v2.20? Simon has seen it; but not sure about when).

izak, 2018-07-05:
Yes, Simon mentioned this to me. I haven't had time to look into it yet. It is supposed to start recharging (to 3%) whenever it reaches 0%. The batterylife code hasn't been touched in a while, so it is unlikely that it is fixed yet. It's also sort-of hard for me to see how it can happen:

```
    def _discharged(self):
        if not self.sustain and (self.soc > self.switch_on_soc or self.soc >= 100):
            return State.BLDefault
        elif self.soc <= 0 or self.soc < self.minsoclimit - Constants.LowSocChargeOffset:
            return State.BLLowSocCharge
```

If the soc is zero, and the top part of the if doesn't execute, it has to go into the LowSocCharge state. There is also a unit test to make sure it operates correctly. So this might not be an easy one to hunt down, if real it's a corner case of some kind, and so far nobody else reported anything either.

SimonHackett, 2018-07-06:
I can aim to re-prove this one but it seems to definitely be happening, at least for me :)

One observation is that we wind up at 0% SoC with no recharging happening, and then if MinSoC is raised to 10% it immediately commences the Recharge mode.

It might be some interaction made more 'interesting' because of our batteries but in principle either its trying to recharge or it isn't... I am wondering a little about a potential interaction with the 'Sustain' function also. Which might be completely wrong!

izak, 2018-07-06:
I tend to think about such things when I go to bed at night, and I started wondering: Are you sure it was all the way down to 0%? The reason I ask is that the GUI rounds the number, so 0.4% will show 0% on the GUI, but it isn't quite low enough for recharge to kick in. Do check it on dbus-spy to be sure.

SimonHackett, 2018-07-06:
Interesting. That’s ‘possible’… but… I also think that you only use the ‘low res’ CANbus SoC option right now - which is only an integer SoC reading anyway…There’s a ‘high res’ one in the CANBus protocol and we send it - but I’m not sure if you actually process it.

Anyway - off for a couple of weeks vacation (yay!) - will return to this when I get back.

SimonHackett, 2018-07-06:
Ok, so it is still there in V2.20~37 (latest release candidate). I just reproduced it again 'at will'.

Scenario:
- set Min SoC to 5%
- let my system (my house system - 2 x 10kWh ZCell) go completely flat overnight - verified the SoC is exactly 0.0%
- observe that no recharging attempt is happening. Venus screen says 'ESS#1' and Bulk mode, but not charging.
- raised MinSoC to 10%
- System then immediately moves to display 'Recharging' and starts to recharge.

See attached images that tell the story above in pictures. I can reproduce this for you any time you want :)

Site example here is https://vrm.victronenergy.com/installation/6208/overview

[image]
[image]
[image]
[image]
[image]
[image]
[image]

izak, 2018-07-13:
You're using `Optimised (without batterylife)`. I'm just going to assume that, because it just smacked me in the face when I looked at that part of the code, but if not, let me know.

When running without batterylife, there was indeed a bug that it would not start recharging at zero.

In fixing it I realised there is another obstacle that prevents us from making the behaviour exactly the same as with batterylife.

With BatteryLife there is a built-in ratcheting hyteresis, when the battery becomes discharged the active limit is raised by 5%, plus an additional 3% hysteresis. This means that:

  * Recharge takes the battery to 3% and then stops
  * Self-consumption resumes at 8%, presumably after the sun is up again.

Without BatteryLife, we go back to self-consumption the moment the SoC rises by 3%. So the effect was that it would recharge to 3% and then immediately start discharging again. Not what we want.

My solution is to recharge to a value lower.

  * When 0% is reached, go into recharge. Charge to 2% SoC.
  * When SoC reaches 3%, go back into Self Consumption mode.

This preserve the spirit of auto-recharge without introducing weirdness.

SimonHackett, 2018-07-13:
You are correct, in terms of the setting we are using - that’s the preference for our sort of battery technology.

Well done for tracking it down...!

---

mpvader, 2018-07-13:
Hey Izak, I'm not sure I can completely follow: in your solution, a system is configured with minimum SOC limit of 5% will only recharge to 3%, right? I would expect it to recharge to 5%. A bit confusing.

Also, @SimonHackett his complaint is that its doing nothing. While you are saying that the bug is that it will charge from 0% to 3%, and then discharge again, etcetera.

izak, 2018-07-13:
My apologies, I am not being completely clear. Let me first discuss Simon's case. If the MinSoc is set to 5%, it will recharge to 5% and go back to the Discharged state. No issues there. But in fixing this I created a new problem. I will discuss that at the end.

First thing to note: There is a static 3% hysteresis. The Soc has to rise 3% above the active SOC limit before it switches back to self consumption.

Second thing: There is a special case for when someone sets the MinSoc as 0%. You cannot exactly recharge to zero, you're already there. So what it does is recharge to 3% and then go into discharged mode, as to not leave the battery completely flat. This was implemented only for WITH batterylife, not for without. That was the bug Simon ran into. 0% wasn't far enough below 5% to activate recharge and without the specual 0% handling it would never activate.

Now for the tricky bit. I had decided back in the day to use the same constant that we use for the hysteresis for the 0% recharge corner case. That is why it is also 3%. BUT, because the without-batterylife side does not have the extra added 5% mentioned in point 1, it means that 3% is also the point where it goes back to self consumption. So it would end up in indefinitely cycling between 0 and 3%.

In short: The fix for simon's MinSoc=5% issue created a new MinSoc=0% issue by using a smaller recharge value than 3% for this corner case.

With all that said though: Isn't there a case to be made for leaving the battery completely flat? Is all this fancy 0% recharging not doing something unnecessary that is actually counter-intuitive?

izak, 2018-07-13:
I deleted the previous comment. I overcomplicated the fix. I'm going to back it out and only handle recharging to the MinSoc for those cases where The MinSoc is actually set to something larger than 0. That fill fix Simon's case and still allow someone to leave the batteries dead flat by setting MinSoc=0%. Which is a perfectly valid use case.

SimonHackett, 2018-07-15:
Great. Yes we definitely want to retain the ability to wind up at dead flat / zero SoC with our batteries.

(It’s their happy place)

Cheers,
  Simon 

---

izak, 2018-07-16:
Hi @SimonHackett ,

I'm happier with this fix. This allows going to complete zero in both optimised modes (with and w/o BatteryLife), but there is a safety catch that activates recharge at 0% IFF MinSoc>=5%.

This issue will be closed when it's included in Venus.

SimonHackett, 2018-07-16:
That sounds like the right sequence (and fix) - will be happy to try it out :)

closed 2018-07-18 by jhofstee