victronenergy/venus #290: Generator running? Disable ESS feedins temporarily

mpvader, 2018-07-05:
Both mppt feedin and Fronius zero feed in

SimonHackett, 2018-07-05:
When a generator is running as the active energy source, the system runs in 'Keep Batteries Charged' mode. 

This request is to ensure that, in addition, we also do these two things (only while generator is on and while it is the active ac source), to further mitigate against any potential to try to 'back feed' energy into the generator:

1) Temporarily engage Fronius Zero-Feed-In
2) Temporarily disable solarcharger Feed-In

These

izak, 2018-07-26:
Presently overvoltage feed-in is already disabled when a generator is present on the active AC input, except in one case: When DVCC is enabled, it will attempt to power the loads from solar once the batteries are full. Am I understanding this correctly that we no longer want this?

mpvader, 2018-07-26:
Well, unless there is a good reason for something, I indeed want to keep the differences between DVCC and not DVCC minimal. So, this difference you mention just now sounds like an undocumented difference?

There are two settings in the GUI: one is Fronius Zero-Feed-in, and the other is Feed-back excess solar power in to the grid.

What I want is the system to behave, when a genset is running as if the Fronius Zero-Feedin is activated, and as if the Feed-back excess solar power into the grid is de-activated.

For the later, its not feed-in 😄 . Feed-back/feed-in, overvoltage feed-in, its still confusing.

mpvader, 2018-07-26:
Ps. since there is a generator running, feeding solar power to the AC loads is nice, but not very important. So if easier (and therefor better to maintain, more stable, etc), just disable overvoltage and be done with it.

The generator will (should) automatically stop once the batteries are full anyway.

izak, 2018-07-26:
There isn't really an *extra* difference between DVCC and non-DVCC. One of the existing known differences are that with DVCC you can power the loads, without it you cannot. This difference simply carries over to generator use.

I am however inclined to agree with you to make it simpler and disable the feeding in of the excess. I anticipated that you might answer in the positive and went ahead and implemented it that way.

Regarding disabling the PV-inverter, that is 100% clear and easy to do.

mpvader, 2018-08-29:
Scheduled for v2.21; part of a larger hub4control change.

closed 2018-10-20 by jhofstee