victronenergy/venus #292: Replace Qt VNC with x11vnc

mpvader, 2018-07-12:
End of 2017, @jhofstee has been experimenting with using vnc from X11, but then without X11.

`x11vnc -noipv6 -rawfb /dev/fb0 -pipeinput UINPUT:direct_key=/dev/input/event0,nouinput -shared -forever`

Advantages:
* multiple clients at the same time
* (probably) having lower cpu cost than the Qt build in one we use today
* (probably) no longer send full frames all the time
* this opens the route to using a newer Qt / QtQuick version, including vnc-ing a display that has been drawed using hardware accelaration

First trials work, to be continued. The todo at least includes:
* Fix keyboard or mouse input
* Security

jhofstee, 2019-11-04:
x11vnc can now be installed with `opkg install x11vnc1` as an optional package on a ccgx. There is no websocket support nor password support. So only install that in a safe environment.

jhofstee, 2024-03-07:
closing this, since:
- https://github.com/jhofstee/qneatvnc/tree/master can add VNC to qt 6 / qml if we want to
- there is gui-v2 with wasm support instead of VNC
- we can make a gui-v1 wasm build as well if want to instead of VNC

closed 2024-03-07 by jhofstee